ArduPilot/ardupilot #9504: RangeFinder: LidarLightV3 I2C update rate slow?

rmackay9, 2018-10-03:
This is slightly anecdotal but while testing a LidarLightV3 I have noticed that the update rate shown in the Ground Station, although accurate, seems to update very slowly as compared to the LidarLightV2.

I performed back-to-back tests.  The VCC, GND, SDA and SCL lines were connected directly to the Pixhawk's I2C port similar to what is shown [here on the wiki](http://ardupilot.org/copter/docs/common-rangefinder-lidarlite.html#connecting-to-the-pixhawk-via-i2c).  A wiring diagram is also shown below although in my case I did not use a BEC.  Still, I suspect skipping the BEC is not the cause of the slow updates.

[image]

rmackay9, 2018-10-04:
This PR may fix the problem: https://github.com/ArduPilot/ardupilot/pull/9169

muramura, 2018-10-04:
I think that the delay time is not used.
In order to use the delay time, I think that it is necessary to set bit 5 of register 4 to 1.
I think that it is now 10 Hz.

[image]

[image]