ArduPilot/ardupilot #9509: Issues building and installing Arducopter on Intel Aero

JorgeG30, 2018-10-03:
### Bug report

**Issue details**
Hello, 

I am running into an issue when trying to build arducopter for the intel aero. I download the source code and follow all the instructions found online. The first issue I notice is when configuring Arducopter for the aerofc-v1. For the NEED_CMATH checks it says that I don't have them. It also says endian.h and byteswap.h are not found. Once I use the command ./waf copter --upload, it makes all the code. The errors arise when exporting NuttX for aerofc-v1. I've attached a text file with the errors produced when trying to upload arducopter to the flight controller. The errors are all similar. They all say something along the lines of 

`error: expected '=', ',', ';', 'asm' or '__attribute__' before '_PARAMS'
 extern double frexp _PARAMS((double, int *));`

I don't know why this is happening or how to fix it. Does anyone have any advice. Thank you

[errors.txt](https://github.com/ArduPilot/ardupilot/files/2444165/errors.txt)

**Version**
Latest version of Arducopter available.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter frame on Intel Aero

**Hardware type**
Intel Aero flight controller

**Logs**
_Please provide a link to any relevant logs that show the issue_

arjun7965, 2018-10-05:
Hi Jorge,
I tried building the fw for aerofc-v1 and did not get any errors. If this is your first time building arducopter and if you are using Ubuntu Xenial try running 

> ./Tools/scripts/install-prereqs-ubuntu.sh

This installs all the dependencies that you need to build the fw.

JorgeG30, 2018-10-06:
Hey Arjun,
Thanks for the information. I actually tried this already. I’ve been following the steps that are found on the ardupilot wiki. I’ve tried building it multiple times already and still get the same errors. I am running Ubuntu like you mentioned.

arjun7965, 2018-10-08:
Jorge,
Can you run me through the steps you did to build the fw? Also, did you try building it for other boards? If yes, do you get the same error?

JorgeG30, 2018-10-08:
The steps I've followed are below. I got them from the ardupilot wiki

`git clone https://github.com/ArduPilot/ardupilot`
`cd ardupilot`
`git submodule update --init --recursive`

After I've cloned the repository I run the prereqs script
`Tools/scripts/install-prereqs-ubuntu.sh -y`
`. ~/.profile`

Once that's done I configure the Arducopter for my board and try uploading
`./waf configure –board aerofc-v1`
`./waf copter –upload`

That's where I get the error. It keeps getting the error at the "generating aerofc-v1.export' part. Oddly enough, I built it for the aero board and it worked perfectly fine. When configuring for the aero board none of the NEED_CMATH checks failed which is weird since they fail each time for the aerofc-v1 configuration

arjun7965, 2018-10-08:
I'm not sure what the problem is. Can you trying doing the following steps:

> ./waf distclean
> ./waf configure --board aerofc-v1
> ./waf copter
> ./waf --targets bin/arducopter --upload

JorgeG30, 2018-10-08:
It doesn't get past the `./waf copter` command. The errors start arising once NuttX is being exported for aerofc-v1. I keep getting the same errors I mentioned before

JorgeG30, 2018-10-12:
Update:
After reinstalling wiping the disk and reinstalling Ubuntu, I am not getting these errors anymore. Now my only issue is just not being able to login to root@intel-aero.local. There were a few warnings when building however.

JorgeG30, 2018-10-12:
Final Update:
I solved the problem. The sshd_config file had to be modified in order to allow root login. This is not recommended due to security concerns but thanks to that I was able to upload arducopter to the flight controller

closed 2018-10-12 by JorgeG30