ArduPilot/ardupilot #9510: Documentation issue on the simulator page

jonathan84clark, 2018-10-03:
On the page
http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html
In the "Start SITL simulator" section the instructions will not allow you to successfully start the simulator if followed. I suspect that this is outdated documentation and simply needs to be updated.
The new way to start the simulator is found in the following forum:
https://discuss.ardupilot.org/t/sim-vehicle-py-no-command-found/21728

Instead of launching the simulator script from the ArduCopter, ArduPlane, etc directory you simply launch it from Tools/autotest/sim_vehicle.py and pass in the vehicle type as a parameter.

This isn't very clear in the documentation, I recommend it be reviewed.

peterbarker, 2018-10-04:
On Wed, 3 Oct 2018, Jonathan L Clark wrote:

> http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html
> In the "Start SITL simulator" section the instructions will not allow you to
> successfully start the simulator if followed. I suspect that this is

What error do you get?

> etc directory you simply launch it from Tools/autotest/sim_vehicle.py and
> pass in the vehicle type as a parameter.

That's how I usually run it...

> This isn't very clear in the documentation, I recommend it be reviewed.

Note that you can fix the documentation via creation of a pull request - 
there's a "Edit on Github" link in the top-right-hand-side of the page.

jonathan84clark, 2018-10-04:
Here is what I am seeing. The documentation does not match up with how you really start the simulator. 
There are aspects of it that are similar but if I am a beginner and I follow the steps in that document exactly, I won't get there. Here is what will happen:
Following Steps in "Start SITL Simulator"
```
cd ardupilot/ArduCopter
sim_vehicle.py
command not found
```
Okay, well lets say I have a little more knowledge about linux and python
```
cd ardupilot/Arducopter
python: can't open file 'sim_vehicle.py': [Errno 2] No such file or directory
```

Then I do an ls into that directory and I realize that the file you indicate is there is not. I check gihub just to make sure my ardupilot repo is correct and I find that it is.
I dig around a little (a.k.a grep) and find it in Tools/autotest/sim_vehicle.py
What's it doing there? How does it know what I want to test
Then I am on the forums trying to figure out how this thing is supposed to work.

The point is that the tutorial on http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html is not accurate on how you actually get a simulator running. And for a someone new to ArduPilot like me it makes it more difficult for me to get things going. I am of the mind that if you are going to have a tutorial on how to get something setup it better be pretty much 100% accurate for someone to follow step by step. 

Does that make sense?

khancyr, 2018-10-04:
Hello,

Did you have a debian based system ? if ubuntu, which version it is ? 
On those system, the step are good as the install script will add sim_vehicle.py to your PATH : http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#install-some-required-packages
Many something else is broken

jonathan84clark, 2018-10-04:
I am using Ubuntu 16.04 LTS. It is possible something is broken. Another things is that I need to have multiple versions of ArduPilot on my Linux machine for different purposes. Each of the versions of the ArduPilot source are stored in different folders. If the setup script is assuming that the path to my ardupilot directory is in my home directory then that path will get setup wrong.
That could be why I am seeing issues. I was trying to get the simulator working with my own code.

khancyr, 2018-10-04:
The script get the real path, but it is done only once. So if you change the path or add new directory, it will not be update. Check with echo $PATH that you get everything. 

I will update the instruction to mention that the direct call to sim_vehicle.py is only possible if the path is correct. 

I am closing as it is solved.

closed 2018-10-04 by khancyr

peterbarker, 2018-10-04:
On Thu, 4 Oct 2018, Jonathan L Clark wrote:

> Here is what I am seeing. The documentation does not match up with how you
> really start the simulator.
> There are aspects of it that are similar but if I am a beginner and I
> follow the steps in that [...]

The instructions in 
http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#add-some-directories-to-your-search-path-facultative 
should have you adding a component to your $PATH such that sim_vehicle.py 
can be run without a path specified.

Can you run "echo $PATH" and paste in the results?

Also "tail $HOME/.bashrc"

> going. I am of the mind that if you are going to have a tutorial on how to
> get something setup it better be pretty much 100% accurate for someone to
> follow step by step.
> 
> Does that make sense?

Absolutely!  Any time someone has problems like this we need to improve 
the documentation.