ArduPilot/ardupilot #9519: Optical Flow Scaler parameters inverted

jmachuca77, 2018-10-05:
### Bug report

**Issue details**
The optical flow FLOW_FXSCALER and FLOW_FYSCALER parameters are inverted, modifying FLOW_FXSCALER changes they OF.flowY values when it should the OF.flowX values instead. Because the scaler is so small only 20% max the behavior is not easily observable. Changing the X value from -200 and logging, and then changing it to 200 and logging exposes the problem.

**Version**
AC3.6rc12 but probably also an issue with older releases.

**Platform**
[ X ] Copter

**Airframe type**
Not relevant

**Hardware type**
The Cube + 3DR PX4Flow + Lightware SF10A

**Logs**

For these logs FLOW_FYSCALER remained at 80, and only FLOW_FXSCALER was modified.

Link to logs:
https://www.dropbox.com/sh/23k1nw6aqasxnim/AADzkVGuyJ3EnqCmZ8WOgfkua?dl=0

This is with FLOW_FXSCALER = -200 --> note OF.flowX is slightly larger than OF.bodyX and OF.flowY is smaller than OF.bodyY

[image]

This is with FLOW_FXSCALER = 200 --> note there was no change on OF.flowX, but now OF.flowY is larger than OF.bodyY

[image]

tridge, 2018-10-06:
the code looks correct to me. I think the explanation is that you have a FLOW_ORIENT_YAW of -9000, which is a 90 degree rotation. The yaw is applied after the scalers, which results in the effect you are seeing.
The reason yaw is applied after scalers is the scalers are a sensor property, not an installation property. We should have better documentation of this however.

jmachuca77, 2018-10-06:
Yeah, I looked at the code also and couldn't find any obvious issues. I can update the wiki to reflect this... but I think the scalers should match the axis don't you? Otherwise you end up with parameter names that don't match the actual functions, and it will get confusing. 
I assume that 90, and 270 will switch the parameters? and 0, 180 will stay the same, possibly 180 and 270 will invert the polarity of the value? but what happens when they are not multiples of 90 deg? say you want to mount it at 45 deg for some reason?

jmachuca77, 2018-10-06:
Also, what about the position offsets? are those also applied like the scalers? or are those correct in relation to body frame?