ArduPilot/ardupilot #9523: Bug in AP_Baro_BMP280, which may or may not harm

olliw42, 2018-10-07:
### Bug report
this is technically a bug, but may not harm thx to the compiler:

_t1 and _p1 are unit16_t, in accordance with datasheet, but are read out with int16_t conversion:

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Baro/AP_Baro_BMP280.cpp#L89
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Baro/AP_Baro_BMP280.cpp#L92

should be
 _t1 = ((uint16_t)buf[1] << 8) | buf[0];
 _p1 = ((uint16_t)buf[7] << 8) | buf[6];

compare also with code in https://github.com/BoschSensortec/BMP280_driver:
https://github.com/BoschSensortec/BMP280_driver/blob/master/bmp280.c#L618-L657

also the (int16_t) conversion for the LSB could be worthwhile to add

7princekumar, 2018-10-07:
I would like to work on this issue. Please assign me.

hendjoshsr71, 2021-09-16:
Closing as it appears to be computed correctly. 

`_t1, _p1` are both defined as `uint16`

See the code here: 
https://github.com/ArduPilot/ardupilot/blob/119df09c448573ac3bf7c7ed766c52fbeb9fd9d1/libraries/AP_Baro/AP_Baro_BMP280.h#L41-L43

closed 2021-09-16 by hendjoshsr71

olliw42, 2021-09-16:
LOL, you didn't got the topic of the PR, right

it's not about how _t1, _p1 are defined, but about how buf is promoted in the conversions, for these two buf is promoted to int16_t and NOT uint16_t (note that for the others promoting buf to int16_t is correct)
please carefully compare to https://github.com/BoschSensortec/BMP280_driver/blob/master/bmp280.c#L698-L721
especially the two lines for _t1 and _p1

so, yes, should be re openend, and just corrected

olliw42, 2021-09-16:
if it helps, here a PR https://github.com/ArduPilot/ardupilot/pull/18682

hendjoshsr71, 2021-09-16:
Yes, I'm aware the issue was about the promotions and importantly about the implicit conversion happening at the very end due to how `_t1` is defined. Where uint16 `_t1` gets set from foo[int16] that has been implicitly converted to uint16 beforehand. 

Thus as the title here says "may not harm."

Thanks for the PR, cleaner and clearer code that doesn't leave it to the coder to figure out the implicit conversions is always better IMO. 

https://godbolt.org/z/TKsoj3MMK