ArduPilot/ardupilot #9536: SITL: Lat/lon converted from degrees to float using single precision, leading to rounding errors

krisklau, 2018-10-09:
### Bug report

**Issue details**

I was working on getting the simulated state out from the SITL instance, and I am using the already implemented FlighGear output stream. However, when using the data I noticed behaviour on the lat/lon output consistent with the typical bug of representing lat/lon with single precision. After digging, I found the error was in conversion from degrees to radians that happens here: 

https://github.com/ArduPilot/ardupilot/blob/014022de03791078f547fd6628c05f90af3229f1/libraries/AP_HAL_SITL/SITL_State.cpp#L251

which uses (as far as I can tell) the conversion routine here:

https://github.com/ArduPilot/ardupilot/blob/014022de03791078f547fd6628c05f90af3229f1/libraries/AP_Math/AP_Math.h#L163

which will result in a degraded output. Locally computing the conversion resolves the issue. 

**Version**
Copter 3.6

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Not relevant

**Hardware type**
SITL

**Logs**
Not relevant

khancyr, 2018-10-09:
Hello,

Thanks for reporting that! SITL is quite a mess with all double/single precision calculation. DEG_TO_RAD_DOUBLE could be use safely on that case ! Would you make a Pull Request to correct it ? It could be good to check other backent on SITL for the same improvement !

krisklau, 2018-10-09:
Sure!

Upon a quick search, I could only find one more occurrence:
https://github.com/ArduPilot/ardupilot/blob/014022de03791078f547fd6628c05f90af3229f1/libraries/AP_HAL_SITL/sitl_gps.cpp#L1230

(searched with grep in AP_HAL_SITL and SITL, quick manual scan for lat/lon references. So might well be more. )

khancyr, 2018-10-09:
we don't want double calculation on vehicle stuff as most hardware currently don't use it. The double should be available only for SITL backend and for thing that are going out.
Anyway, just make a PR with the change for your problem will be enough for now !

krisklau, 2018-10-22:
I submitted the PR :)

OXINARF, 2019-02-02:
Fixed by #9541.

closed 2019-02-02 by OXINARF