ArduPilot/ardupilot #9552: Create AP_Telemetry Library

auturgy, 2018-10-12:
### Feature request

There are a number of data protocols already supported, and also a significant number of requests for alternative protocols.  Abstracting how ArduPilot consumes and generates telemetry/data/commands will make it easier to manage both the existing and requested/future protocols.
This issue is raised to aid tracking, and suggest a flexible way forward.

**Describe the solution you'd like**
Creation of an AP_Telemetry library to enable frontend/backend split for telemetry/data/commands

**Describe alternatives you've considered**
Continue adding new protocols in their own libraries and/or through AP_SerialManager

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Related Issues**
iBus: https://github.com/ArduPilot/ardupilot/issues/9419
LTM: https://github.com/ArduPilot/ardupilot/issues/3193 & https://github.com/ArduPilot/ardupilot/issues/9469
NMEA: https://github.com/ArduPilot/ardupilot/issues/1774
CastleLink: https://github.com/ArduPilot/ardupilot/issues/2218 
HoTT: https://github.com/ArduPilot/ardupilot/issues/856
GSM Modem: https://github.com/ArduPilot/ardupilot/issues/1641
High Latency: https://github.com/ArduPilot/ardupilot/issues/7341
FrSky: https://github.com/ArduPilot/ardupilot/issues/5725 & https://github.com/ArduPilot/ardupilot/issues/6296 & https://github.com/ArduPilot/ardupilot/issues/8922 & https://github.com/ArduPilot/ardupilot/issues/4812

**Related Pull Requests**
https://github.com/ArduPilot/ardupilot/pull/7819 [Devo Telem, merged]
https://github.com/ArduPilot/ardupilot/pull/6325 [MQTT, closed]
https://github.com/ArduPilot/ardupilot/pull/957 [HoTT, closed]

Saijin-Naib, 2018-10-23:
I think this is an incredibly valuable idea and direction to take AdruPilot, and I'm really thankful you've compiled all these requests from various users, including myself.

peterbarker, 2018-10-26:
On Thu, 25 Oct 2018, Luis Vale Gonçalves wrote:

> @auturgy you should also add MAVLink as a telemetry backend.
> 
> Also should be added those telemetry protocols that support input

I'm pretty sure the current `gcs().` can form the basis of this telemetry 
library.

Saijin-Naib, 2018-10-29:
Guys, I think this PR should be included as well: 
Support for Tramp telemetry/smart audio VTx/Cam control over serial port
https://github.com/ArduPilot/ardupilot/issues/9565

davidbitton, 2019-01-10:
How about Jeti EXBUS which has RC in one direction and telemetry in the other. One connection between the RX and the FC.

mboland, 2019-01-11:
+1 for JETI Telemetry. Currently doing it with an Arduino board.
JETI is becoming a much more widely used product due to its quality and sophistication

davidbitton, 2019-01-11:
they also have a new radio at a much lower price point.

andyp1per, 2020-04-18:
This contains the beginnings of this.

closed 2020-05-04 by tridge