ArduPilot/ardupilot #9553: Rover: PWM_TYPE_BRUSHED_WITH_RELAY does not work after reboot if output channel is reversed

fabio-d, 2018-10-12:
### Bug report

**Issue details**

If PWM_TYPE is BrushedWithRelay and one or more output channels are configured as REVERSED, **motors on such channels start spinning at very fast constant speed** if the board is power cycled and there is no way to control them or stop them.

Examination of the parameters' values shows that **the TRIM value is overwritten and set to the same value as MAX at boot** (which means that, since in BrushedWithRelay mode the PWM output is always in range [TRIM, MAX], the motor will always run at TRIM=MAX speed).

This behaviour only happens if the output channel is REVERSED. The non-REVERSED channels keep working fine even if the board is power cycled.

**Version**
ArduRover 3.4.2

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Airframe type**
Skid steering platform, each of the two motors controlled through a driver that takes a PWM line and a direction line as input. I have configured PWM_TYPE=BrushedWithRelay and SERVOn_{MIN/TRIM/MAX} values (with TRIM=MIN=800 and MAX=1900), SERVOn_FUNCTION = ThrottleRight and ThrottleLeft. I also connected the direction lines to the Relay pins (AUX 5 and 6).

Because of the polarity of the left motor, I also had to reverse the left channel (SERVO3_REVERSED=1). This setup worked perfectly (=I could successfully manoeuvre the rover over RC) until I rebooted the board, then the left side started exhibiting the behaviour described above.

**Hardware type**
Pixhawk (px4-v2)

fabio-d, 2018-10-12:
Further investigation seems to show that the culprit is the `set_trim_to_min_for` call in `AP_MotorsUGV::setup_safety_output()`, which indeed sets the TRIM value to MAX if REVERSED=1, or MIN if REVERSED=0.

I wrote a patch that should solve the issue by always setting TRIM to the MIN value, regardless of the REVERSED flag (https://gist.github.com/fabio-d/d6a9a860d47a6e8651e543b97944f428) and I confirm that it works for me. Please note that I created a separate `set_trim_to_absolute_min_for` function to avoid interfering with Plane, which also uses `set_trim_to_min_for`.

hdtechk, 2018-10-12:
You do know that you can reverse brushed motor by swapping pos. Neg wires

fabio-d, 2018-10-12:
Hi @hdtechk, thanks for your reply! Yes, I was aware of that. The thing is, I'm trying to repurpose an old rover by replacing its proprietary logic board with a Pixhawk to get all the extra features and, ideally, I wouldn't want to have to rewire anything else. In this regard, the REVERSED parameter fits my need perfectly.
I would also add that the current behaviour of the REVERSED parameter, which causes the motors to start spinning at full speed after a reboot, is possibly dangerous and surely unexpected! :wink:

IamPete1, 2021-09-15:
Fixed by https://github.com/ArduPilot/ardupilot/pull/18494

closed 2021-09-15 by IamPete1