ArduPilot/ardupilot #9559: Compass Calibration Pre-Arm Check with saved parameters

Naterater, 2018-10-14:
### Bug report

**Issue details**

I saved my parameters after completing a compass calibration. After changing firmwares and parameters, I came back to the same firmware, I uploaded the correct calibrated parameters, but I could not arm until I actually ran a compass calibration again. 

**Version**
Plane 3.9.2

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Traditional electric

**Hardware type**
Cube Black, HERE+ GPS/Compass

**Logs**
Will try to dig up if requested

Pedals2Paddles, 2018-10-20:
I don't think it will all that. Always wants a recalibration after a reset. I'm not sure why, but I've always found it to be annoying.

IamPete1, 2021-08-29:
We don't allow this to try and stop people loading calibration values from different vehicles. 

The bypass is now documented in the wiki:
https://ardupilot.org/plane/docs/common-parameter-reset.html#restoring-calibrations-after-resets

closed 2021-08-29 by IamPete1