ArduPilot/ardupilot #9563: AP_Airspeed_UAVCAN: Bug in temperature reading

olliw42, 2018-10-15:
it should read differential_pressure_sensor_temperature , not static_air_temperature 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Airspeed/AP_Airspeed_UAVCAN.cpp#L128-L142

what should actually happen is that AP_Baro also registers to the RawAirData message, and reads out the info the AP_Baro class is related to, e.g. static_air_temperature

EShamaev, 2018-10-15:
As for temperature, it's not used anywhere, but thanks for noting, will fix.

WickedShell, 2018-10-15:
Airspeed temperature is used for barometric height calculation of available.

EShamaev, 2018-10-15:
In AP_Airspeed?

EShamaev, 2018-10-15:
Looked again through the code, if it is used in the meaning of air temperature - then there is no bug, all is correct as it is.
@WickedShell please confirm the actual meaning of "temperature" in AP_Airspeed.

olliw42, 2018-10-15:
with the current concept and handling in AP it is a bug in any case

the RawAirData explicitly introduces three temperatures, and I think it is clear what each is supposed to mean.  static_air_temperature is certainly not associated to the diff_press sensor. For your convenience I reproduce what the standard says:

float16 static_air_temperature          # Kelvin
                                        # This field contains the raw temperature reading 
                                        # from the externally mounted temperature sensor or, 
                                        # in absence of one, the raw temperature of the pressure sensor.

furthermore, going through the drivers for the other airspeed sensors it is clear that temperature reported there is the one the sensor reports, see e.g. MS5611 as an example.
thus, the above conclusion is the only one which is consistent.

EShamaev, 2018-10-15:
I disagree until we find out what this temperature is used for.

If it is used in calculations of air density parameters, then all is fine now.
If it is used for corrections of sensor values depending on temperature of sensor, then needs to be fixed.

EShamaev, 2018-10-15:
As for "consistency" - look here:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Airspeed/AP_Airspeed_SDP3X.cpp#L232

In this driver it is used for air density calculations and corrections.
If we take this example - then it should be outside air temperature and not sensor temperature.

EShamaev, 2018-10-15:
Well, all I can say is that as always there is only one right point of view, which of course is yours.

As @WickedShell wrote, this temperature is used for barometric height calculation if available. So the logical measurement for that is external temperature sensor which is real air temperature.
And as perfectly stated in standard - if there is no outside sensor - temperature of pressure sensor is transmitted in this field. Which again, perfectly matches the AP logic and brings us to the case of other sensors already implemented.
So you are right, standard is clear.

I was not the author for AP_Airspeed_UAVCAN but what was implemented is correct from the AP and UAVCAN point of view.

>  I would expect that you also should require the UAVCAN node to do the air density calculations and corrections, if possible, e.g if the static fields are populated

I would not. I would expect the node to transmit messages 1020 and 1021 and AP would not care about raw air data.

closed 2018-10-15 by EShamaev