ArduPilot/ardupilot #9564: SBUS channel deranged on FMUV5

jamming, 2018-10-15:
#### Issue details
I connected a futaba or frsky  sbus  RC to the SBUS-port of  PX4FMU-V5,  and make the RC working in D16 mode.  After Radio Calibration all Radio channels work well. But if I move the joystick constantly for several minutes or more, the Radio channels deranged occasionally. Once that happens, the Radio channels jitter randomly even the  joystick is stoping. When reset or re-power the FC, this issue disappear.  But it will happen again if I constantly moving the joystick for enough time.

#### Version
ArduCopter V3.6.0-rc12

#### Platform
[*] Copter

#### Airframe type
QuadX

#### Hardware type
PX4FMU-V5

lvale, 2018-10-15:
A log is required to analyse the issue.

jamming, 2018-10-25:
See in the log file

lvale, 2018-10-25:
You mention that you connected to the SBus port, but you don't say which specific model of controller.
This specific one has no problems and the radio is connected to the RC-IN port 
[image]

tridge, 2018-10-25:
by "joystick" do you mean the sticks on the transmitter, or do you have a separate joystick connected? What TX are you using? What RX model?

auturgy, 2018-10-25:
D16 mode implies FrSky X series (Taranis).

Regards,

James

jamming, 2018-10-30:
I use Pixhawk4 controller and  Futaba T8FG.  The radio is connected to the DSM/SBUS RC port.
When I flash pix4 1.8.0 firmware into the  Pixhawk4 controller, SBUS port works well both D8 and D16 mode.  
But when flashed with ardupilot firmware, SBUS port can not work stable with D16 mode . Only D8 mode is OK.

jamming, 2018-10-30:
Thay are  the sticks on the transmitter.
I have tested Futaba T8FG(with Futaba Fasst compatible RX)  and  frsky X9D(with XSR Rx),  thay all have the same problems.

jamming, 2018-10-30:
I have looked into the circuit of pixhawk4 and found that there are some differences of the Rx circuit. DSM and SUBS share the same connector and the same signal pin on the pixhawk4. So there are signal on PA10 and PB11 at the same time when we connect either DSM or SBUS receiver. I found that radio channels confusion  are most  likely happen  when RX RSSI low or RX singal momentary lost.I hope this feedback will help you solve the problem.

rmackay9, 2019-01-11:
This is fixed in master and Copter-3.6.4 so closing.  Because the Pixhawk4 can't be distinguished from a regular Pixhawk users must specifically download the firmware from the directory listed below and then manually load to the board using the ground stations, "Load custom firmware" option

http://firmware.ardupilot.org/Copter/stable/Pixhawk4/arducopter.apj

closed 2019-01-11 by rmackay9