ArduPilot/ardupilot #9565: Support for Tramp telemetry/smart audio VTx/Cam control over serial port

athertop, 2018-10-15:
### Feature request

**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
Would like for Ardupilot to support the Tramp Telemetry (and smart Audio alternative) for allowing configuration of supported VTx modules (and some cameras like the Runcam Split mini v2). This allows RC control access of inbuilt device menu system over OSD - providing ability to, for example:
- change VTx channel/band, power level etc
- change camera options, resolution, stills/movie etc

**Describe alternatives you've considered**
The only alternative is to use Betaflight where this support is already implemented for both Tramp and Smart Audio devices - As I am an ardupilot-pilot :-) I would really love to see these features available here.

**Platform**
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Overview of Tramp and Smart Audio telemetry can be seen here: https://blog.dronetrest.com/smart-audio-vtx/

Cheers, Paul

andyp1per, 2020-04-18:
RunCam support has been merged into master. This provide OSD access via stick commands

jimpriest, 2020-04-26:
Would love to see this feature as well.  I use my VTX camera while flying but it's a pain to switch VTX power. 

Betaflight has a neat feature where the VTX is in pit mode until you arm then you can go to max power,  and back down on disarm.  

Would love to see this feature in Arducopter as I use my FPV cam as sort of a safety net while flying so I can go to manual mode in an emergency aka  the trees start moving around :)

jimpriest, 2020-04-26:
@andyp1per    Is there any documentation on the stick commands for OSD?  Does that provide what the OP was asking for as far as setting VTX power etc?

andyp1per, 2020-04-27:
@jimpriest I realised that I cannot make this work without OSD support for parameter modification. RunCam is different as it has its own OSD, but for VTX we have to supply the OSD. So I have worked on this: https://github.com/ArduPilot/ardupilot/pull/14203 which seems to work quite well. If I can get it merged I will next work on exposing CSRF and then SmartAudio settings as parameters that could be put on the OSD.

jimpriest, 2020-04-27:
@andyp1per thanks for clarification!  Good luck on your pull request :)

IamPete1, 2021-07-26:
@andyp1per I think we can close this now?

andyp1per, 2021-07-26:
Implemented

closed 2021-07-26 by andyp1per

athertop, 2021-08-08:
Was IRC Tramp support added to AP? I see smartaudio now supported but don't see any mention of IRC Tramp support.