ArduPilot/ardupilot #9566: AP_Airspeed_UAVCAN: Temperature correction c(sh)ould be consistent with AP_Baro

olliw42, 2018-10-15:
the correction of temperature from UAVCAN to AP standard, i.e., K to C, is done in AP_Baro_UAVCAN in the UAVCAN handler, i.e. the driver field is populated with AP conform data:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Baro/AP_Baro_UAVCAN.cpp#L168
In contrast, in AP_Airspeed_UAVCAN the driver field is populated using the UAVCAN convention:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Airspeed/AP_Airspeed_UAVCAN.cpp#L136
and the correction to AP convention done later in get_temperature:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Airspeed/AP_Airspeed_UAVCAN.cpp#L171

Populating already the driver field with AP conform data makes most sense IMHO, and AP_Airspeed_UAVCAN could/should do what AP_Baro_UAVCAN does.

EShamaev, 2018-10-15:
Absolutely right!

EShamaev, 2018-10-15:
Please make a PR for this.

OXINARF, 2018-11-19:
Fixed by #9650

closed 2018-11-19 by OXINARF