ArduPilot/ardupilot #9567: External Temperature Probe Support

micaheli, 2018-10-15:
### Feature request

**Is your feature request related to a problem? Please describe.**
With the growing use of 18650 packs and LiIon packs in general, monitoring temperatures of things is increasingly important.  In addition to component monitoring, external air temperature can be a useful tool for compensation when measuring barometric pressure and airspeed.   

Currently, the accepted solution is to disable your battery monitoring (not really an acceptable workaround) and monitor your temperature in volts.  

**Describe the solution you'd like**
The vision I have for this feature would be to have a couple configurable parameters in order to utilize as well as display on OSD a temperature for an external probe.  Said feature would also comply with imperial/metric global settings.

It seems that all or most of the accepted temperature sensors output 0-3.3v as the scale, so a default scale/min/max could be set for the most common sensors
Examples of potential variables:
EXT_TEMP_PIN <configures the analog pin to use for this reading>
EXT_TEMP_SCALE <scaling variable to calibrate temperature to>
EXT_TEMP_MIN <min voltage to calibrate against
EXT_TEMP_MAX <max voltage to calibrate against

OSD1_ETEMP_EN <enable/disable external temp on OSD>
OSD1_ETEMP_X <X position of element on screen>
OSD1_ETEMP_Y <Y position of element on screen>

**Describe alternatives you've considered**
Using separate telemetry systems such as FrSky s.port to transmit temperature using their temperature modules that already exist.  

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

micaheli, 2018-10-15:
Obviously this should go without saying, but for ANY real-world aircraft operations, temperature is a hugely critical variable that needs to be known before flight and during flight to prevent things like icing of control surfaces and internal combustion engine tuning/efficiency.  

For things like the Rover/Submarine, I would imagine that getting water or soil temperatures could really open up possibilities as well.  

Not sure how this could benefit the AntennaTracker, but it certainly would get use on a display for outside temp.

vierfuffzig, 2018-10-17:
respective options have been added in the last months, so with current code on a suitable controller it is well possible to use an auxilliary I2C barometer (e.g. bmp280 type) for external temperature readings and display them on your onboard OSD using the OSDn_TEMP and _BTEMP items. see https://github.com/ArduPilot/ardupilot/pull/9258 and https://github.com/ArduPilot/ardupilot/pull/9386

additionally, you can use a digital airspeed sensor's outside temperature reading, if existent, and display it using the OSDn_ATEMP item.

prior to these additions, i have been been tinkering with TMP36-type analogue sensors. seemingly ideal for this type of application and easy to attach to a free adc with a just the multiplier and offset to adjust, i have found them to be that heavily prone to external HF interference that i could not get them to provide useful data in a typical FPV-type airframe setting.

cheers, basti.

micaheli, 2018-10-17:
I didn't think you could configure multiple barometers - so relocating a
barometer on a battery (ie, for monitoring temperature) or for any of the
other use cases (other than air temp) is really a moot point.   I need an
external temperature sensor/feature to be able to have an
auxiliary temperature displayed in addition to the currently utilized
sensors.

My main use case here is to put a sensor in a velcro strap that I wrap
around the flight battery in order to ensure that I'm pulling a safe amount
of amps.

Naterater, 2019-01-18:
I think this should be closed in favor of https://github.com/ArduPilot/ardupilot/issues/877