ArduPilot/ardupilot #9570: ChibiOS:  need sensors reset feature.

liang-tang, 2018-10-16:
### Feature request

there is no sensors reset feature for chibios systerm,
nuttx sytterm imple....
if (px4_start_driver(fmu_main, "fmu", "sensor_reset 20")) 

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

IamPete1, 2021-08-29:
Do we need one? What is the use case?

IamPete1, 2021-09-11:
Please re-open if your still interested in this.

closed 2021-09-11 by IamPete1