ArduPilot/ardupilot #9571: Copter: dynamic vertical speed descending

robustini, 2018-10-16:
### Feature request

As is well known when a copter quickly falls on its vertical the propellers can stall, the classic effect that is noticed is an oscillation of the copter.
This phenomenon is absolutely less when the copter doesn't descend vertically, but does so by translating it.
It would therefore be interesting to have a higher modular descent speed when the ground speed is not equal to 0 ".
Example:

PILOT_SPEED_DN_HOVER = 150 cm/sec
PILOT_SPEED_DN_MAX = 300 cm/sec

It may already be sufficient to evaluate the amount of the roll or pitch command to create an automatic modulation of the descent speed, or base to the gruound speed.
The same concept could be applied to automatic flight.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2021-09-07:
@lthall Any thoughts on this issue? I tend to think we should close....

With the correct tune you should be able to descend just as fast no matter if level or not. 

If we were to implement this I think it would be quite unintuitive to have the throttle stick meaning different things depending on airspeed.

lthall, 2021-09-07:
Randy has done a circular decent LUA script that does something similar to what is described here, and the Callisto is able to descend vertically everywhere from 0 to terminal velocity and that is a 50 kg multirotor (By terminal velocity I mean drawing 3A on an aircraft requiring 120A to hover).

closed 2021-09-07 by lthall