ArduPilot/ardupilot #9573: Autotest: The parameters are not cleaned before the tests

Williangalvani, 2018-10-16:
When running the Autotest some parameters are updated from the .parm files, but these are applied over the previous configuration, which means that any parameters not on these files remain unchanged. 
This can lead to inconsistent behavior for different users as some parameters could have been changed during SITL runs, for example.

Shouldn't the EEPROM be cleaned before loading the autotest parameters?

peterbarker, 2018-10-17:
On Tue, 16 Oct 2018, Willian Galvani wrote:

> Shouldn't the EEPROM be cleaned before loading the autotest parameters?

Yes it should, and this directive:

https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/arducopter.py#L96

Should make that the case.  Are you suggesting that's not working?

Peter

Williangalvani, 2018-10-17:
Yes, I had some problems with parameters persisting, but unfortunately now I can't reproduce it (and I'm starting to think it might have been my mistake), so I'm closing this.

closed 2018-10-17 by Williangalvani