ArduPilot/ardupilot #9579: AC3.6.0-rc12 UAVCAN Scheduler Issue?

edwinhayes, 2018-10-17:
### Bug report

**Description**

Flash a cube with RC12 (via QGroundControl).  Connect a Zubax GNSS2 to CAN2 port.  Enable physical port driver and set protocol to UAVCAN.  Set GPS1 type to UAVCAN.  GPS is detected, sees sats and provides a fix (though HDOP is pretty terrible inside on my desk).  But periodically MAVLink status text message "Initialising APM" is published.

The interval between messages isn't regular.  In the range of thirty seconds to five minutes seems typical.

Unplugging the Zubax GNSS2 results in the message stopping, and then replugging it causes the message to start again.

Looks like disabling the GPS (via GPS_TYPE) doesn't stop the message.  Maybe reduces the frequency, but difficult to tell?

If I close QGroundControl, and just watch the telemetry stream passively, the messages still continue, so probably not related to interaction with the ground control software.

From a cursory look, this message is spawned when either a CPU reset or a scheduler issue means that `mavlink_delay_cb` isn't called frequently enough?  Looks like MAVLink sequence numbers aren't interrupted/reset, so I assume the latter.

**Version**
ArduCopter V3.6.0-rc12 (014022de)

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
Cube Black

**Logs**
Attached is a DataFlash log; captured while disarmed on the desk and then downloaded via MAVLink.

[log_46_2018-10-18-10-47-18.bin.zip](https://github.com/ArduPilot/ardupilot/files/2489446/log_46_2018-10-18-10-47-18.bin.zip)

edwinhayes, 2018-10-24:
Further testing seems to indicate this affects the NuttX build only; flashed ChibiOS and it's been running for ~45min now without the issue.

rmackay9, 2018-10-26:
@edwinhayes, thanks for the report.  The performance figures in the log are pretty bad.  It looks like the main loop is freezing up for up to 0.19 seconds at times.  The vehicle would twitch very badly or perhaps even crash.
[image]

@OXINARF, I think you have one of these GPSs, have you seen this issue?

EShamaev, 2018-10-26:
> Further testing seems to indicate this affects the NuttX build only

Do we still support NuttX?

rmackay9, 2018-10-27:
@EShamaev, yes, for Copter-3.6.0 NuttX is still the default but ChibiOS is available for those who want to use it.  From Copter-3.7.0 onward ChibiOS will be the only OS option for the pixhawk family of boards.

OXINARF, 2018-10-27:
@edwinhayes I've tried to reproduce this multiple times, with more and less load on the system and I can't reproduce it. Can you say what you have connected?

edwinhayes, 2018-10-28:
@OXINARF Hmm ok.  It's a black cube on a standard carrier.  TELEM1 connects to a companion computer, then UDP to the ground.  There are quite a few other logical MAVLink components on the companion computer, to the flight controller will see heartbeats from these components, but that's about all.  No other GPS, and I've got the internal compasses disabled for this testing.

I'm running QGroundControl rather than Mission Planner for this testing, because I'm on a Mac - from memory, I believe that by default QGC sets the stream rates on the flight controller higher than Mission Planner does?

Also, I didn't change anything on the GNSS, other than the default node id, so the CAN streaming rates will be whatever was default.

edwinhayes, 2018-10-30:
Hmm, weird.  Ok, once I have a free moment I'll try to flash it back to NuttX again to confirm its still there.

edwinhayes, 2018-10-31:
Flashed to V3.6.0 NuttX, problem still occurs for me.
[image]

rmackay9, 2018-11-01:
To provide another data point on this issue, rickyg32 is reporting on [this Copter-3.6 thread](https://discuss.ardupilot.org/t/copter-3-6-0-available-soft-release-for-2-days/34456/41) that he's seeing "Bad AHRS" and "Vertical position issues" while using this UAVCAN GPS.  His log also shows a single PM message with a longest loop time of 47274 (47ms).  The vehicle has ChibiOS on it.

EShamaev, 2018-11-01:
@edwinhayes 
Please provide full settings list for Zubax GNSS 2.
It is done with <cfg list> command from command line interface over the USB virtual serial port.

edwinhayes, 2018-11-05:
Hmm, curiouser and curiouser.  Previously, I had `GPS_TYPE` set to UAVCAN, and only the Zubax GNSS2 connected.  Now, I have `GPS_TYPE` set to uBlox, with a Here2 connected via UART, and `GPS_TYPE2` set to UAVCAN.  Both GPS are detected, but now the "APM Initialising" messages have stopped, so I think the problem has gone.  Perhaps this is why others haven't reproduced?

EShamaev, 2018-11-09:
I have CAN only plane without any UART GPS. However on plane.
Can you please load the plane firmware just to check?

amilcarlucas, 2021-01-05:
Master has had some fixes for this. Can you re-test?

edwinhayes, 2021-01-06:
@AntonSlooten FYI.

IamPete1, 2021-07-26:
I think the conclusion is that this is fixed now? Please re-open if i'm wrong.

closed 2021-07-26 by IamPete1