ArduPilot/ardupilot #9584: Building Arducopter scripts

JorgeG30, 2018-10-18:
#### Issue details
Hello, I was just wondering how does one build and upload scripts to the flight controller using Waf. I have built and uploaded the examples that come with arducopter but I would like to build my own codes and run them. I am using the Intel Aero flight controller.

#### Version
Copter 3.6

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quadcopter

#### Hardware type
Intel Aero Drone with Flight Controller

#### Logs

auturgy, 2018-10-18:
Please use the forum at discuss.ardupilot.org for support requests, or join the developer chat at gitter.im/ArduPilot 
Re waf, https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md probably has the information you’re looking for.

Regards,

James

JorgeG30, 2018-10-18:
Will do, thanks

closed 2018-10-19 by lvale