ArduPilot/ardupilot #9589: Autotest: Add test badges to README

Williangalvani, 2018-10-19:
Since the autotest is too slow to run on every pull request, why not show it at least on the Readme page? It is important that it shows somewhere else than at http://autotest.ardupilot.org/, so people at least know when something is broken without going there.

As there are 26 tests right now, I propose we either group them all into something like [image] 
Or group by vehicles/task
[image]  [image]  [image]

Any thoughts?

umer936, 2018-10-24:
I like this idea! Did you want to make a PR or can I?

Williangalvani, 2018-10-24:
I haven't done anything, feel free to do it!

PR #10449 (2019-02-12): Add Autotest badge
This adds badges for the Autotest results in the Autotest web interface itself and to our Readme. Hopefully this improves awareness of failing tests there.

Passing:
> [image]

Failing:
> [image]

The [badge](https://img.shields.io/badge/Autotest-1\2-red.svg) is downloaded from the [shields.io](https://shields.io) website into the `buildlogs` folder and is acessible at `/autotest-badge.svg` so we can point at it from our Readme.

PS: The Readme image will be broken until the Autotest runs with these changes.

Fix #9589
closed 2019-02-12 by peterbarker