ArduPilot/ardupilot #9592: Run simple autotests for each vehicle in CI

patrickelectric, 2018-10-19:
I think that it's possible to run small and fast autotests for each vehicle, this would help to identify if a PR breaks a vehicle functionality. 
We should aim to use travis or another third party CI system just to test it (like appveyor).
Right now the full autotest setup takes 2h-ish to finish everything for all vehicles, that idea is to run just simple tests that can check the most basic functionalities of each vehicle.

peterbarker, 2018-10-20:
On Fri, 19 Oct 2018, Patrick José Pereira wrote:

> I think that it's possible to run small and fast autotests for each vehicle, this would help to identify if a PR breaks a vehicle functionality.
> We should aim to use travis or another third party CI [...]

Something I've wanted to do for a while is to name the tests we pass into 
"run_test".  That would allow us to run individual tests by name.  This 
has the prerequisite that each test is self-contained, however - assuming 
a precondition of being on the ground, disarmed.

jaxxzer, 2019-02-05:
I think this can be closed? did we want to run more tests than this?
https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/build_ci.sh#L69

closed 2019-02-05 by patrickelectric