ArduPilot/ardupilot #9612: That's not correct. When EK2_ALT_SOURCE is set to 2 it will use the GPS 100%. While it is doing this, it constantly calculates an offset for the baro alt, so if GPS is lost it can switch to baro without a height jump.

hschioler, 2018-10-23:
_Originally posted by @priseborough in https://github.com/ArduPilot/ardupilot/issues/2022#issuecomment-182326833_

hschioler, 2018-10-23:
The parameter EK2_GPS_TYPE does not allow for 3D pos.
How is GPS then used directly for altitude ?

priseborough, 2018-10-23:
http://ardupilot.org/plane/docs/parameters.html#ek2-alt-source-primary-altitude-sensor-source

closed 2018-10-23 by priseborough

hschioler, 2018-10-23:
I asked because I am trying to inject artificial GPS (NMEA) to the GPS port.
The indoor positioning system used is quite precise - also in altitude.
So assume EK2_ALT_SOURCE=2  how should I set the EK2_GPS_TYPE to ensure GPS pos data are
the primary driver for the EKF2 ?
As you know NMEA does not have a field for vertical velocity.