ArduPilot/ardupilot #9616: Copter climb infinitly if rangefinder is stuck

khancyr, 2018-10-23:
### Bug report

**Issue details**

On Copter, if the rangefinder is stuck on a value (wire problem, fog, leaf, etc) then the copter will climb on descend infinitly in Loiter, poshold, alt_hold modes.

This is due to the call of get_surface_tracking_climb_rate() function that only check the rangefinder heatlth (that isn't a well named function). 

My proposal is to add a check from the EKF or baro to validate the rangefinder measurement or improve the glitch counter to take into account the stucking.

**Version**
copter 3.5 to 3.7-dev
**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2018-10-24:
I think this is a duplicate of this existing issue: https://github.com/ArduPilot/ardupilot/issues/4697

khancyr, 2018-10-24:
No exactly a duplicate. It extends it . 
The fact is that it will append with all rangefinder and without RNGFND_MAX too. The only condition is to have a wrong value in acceptance range.
I will works on this to  add more check on rangefinder for altitude. The healthy() function is tricky as it only check if the sensor is present. We could double check the rangefinder health with EKF, baro and GPS alt

RickReeser, 2018-10-26:
I have found that this can also be caused by the Lightware serial driver misinterpreting the lidar's "no reading" value of -1 as 1 meter. #7756

highfreq, 2018-10-28:
I can confirm it happens in fog with Lightware SF11/C digital lidar. It thinks fog is terrain and keeps climbing. It happened to me last year, it climbed untill it cleared fog at 2100 feet........
Due t fog i couldn't switch to stabilize because it got out of sight. After it got clear of fog it started descending as instructed but really really slowly, probably EKF was thinking it would have touched ground soon due to rangefinder reading fog.

Corrado

chobitsfan, 2019-03-22:
I have the similar problem, my copters climbing up to dangerous 1200m before it starts to decent
https://discuss.ardupilot.org/t/copter3-5-7-the-drone-will-not-rtl-and-landing-always-fly-up/39978

It seems related to #6765

KiloNovemberDelta, 2019-07-26:
This topic seems to have the same behaviour : https://discuss.ardupilot.org/t/tfmini-copter-does-not-stop-climbing/44533/8 
Is there any fix going on ?