ArduPilot/ardupilot #9622: EKF2/EKF3: always push data to EKF instead of having EKF sometimes pull from sensor

rmackay9, 2018-10-24:
In EKF2/3 we sometimes push data into the EKF (for example we push external position estimates into the EKF) but other times the EKF pulls data directly from the sensor (i.e. rangefinder?).

We should consolidate the interface so that the vehicle code (or perhaps sensor itself) is always pushing data into the EKF.

The benefits of doing this are:

- more consistent interface
- allows replay to work better?

I'm creating this issue because I've heard @priseborough mention this a couple of times.

int-smart, 2018-10-28:
Hi. I am looking in this issue. I am not able to figure this out. I have looked into the copter AP_NavEKF3 code and the only difference that I am seeing is that NavEKF3 class has explicitly defined range finder in the class constructor while other sensors are not. The class however has methods that makes call to estimates from sensors such as gyro etc. Can you give me some more information on this and if I am moving in the right code.

int-smart, 2018-11-03:
Hi @rmackay9 Can you help me understand this problem a little more by elaborating the problem or giving some hints as to where in the code should I look into.