ArduPilot/ardupilot #9640: Copter: Compass calibration cannot be initiated if RNGFND_TYPE set to 14, but no rangefinder connected

aercamti, 2018-10-26:
https://discuss.ardupilot.org/t/3-compasses-installed-problems-starting-compass-calibration-enumeration-missing-docs-help-please/33661/22
### Bug report
If the parameter RNGFND_TYPE is set to 14, and the (Teraranger One) rangefinder is NOT connected, then the compass calibration cannot be started. 
(If changing this param to “0”, upon reboot, compass calibration can be initiated as expected.)

Conversely, if parameter RNGFND_TYPE _is_ set to 14, the rangefinder has to be connected. Then compass calibration can be initiated.

The issue is repeatable.
Setup has a Here+ and Here module connected.
**Version**
3.5.7
**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Hexa
**Hardware type**
Pixhawk 2.1 Cube
**Logs**
n/a

rmackay9, 2018-10-27:
@aercamti, could you try Copter-3.6.0 which will be available in the next few hours from the ground station's Beta Firmwares link?  I think 3.6.0 includes a fix to this issue.

IamPete1, 2021-07-26:
No update, @rmackay9 thinks 3.6.0 fixed so I'm closing.

closed 2021-07-26 by IamPete1