ArduPilot/ardupilot #9645: Newer batches of LidarLite V3 have incorrect distance reading

lukedempsey, 2018-10-27:
### Bug report

**Issue details**

I have two different LidarLite V3 modules. One is from an early batch of V3s, one is from a later batch. The early batch module works fine. The module from the later batch always shows a reading of ~1400 (14m). When the sensor is directly covered (hand over lenses) it reads ~940.

Currently, the way the lidar is read from is that it is set to continuous capture, and the distance registers are read from at 50Hz

Setup:
```
static const struct settings_table settings_v2[] = {
    { LL40LS_INTERVAL, 0x28 }, // 0x28 == 50Hz
    { LL40LS_COUNT, LL40LS_COUNT_CONTINUOUS },
    { LL40LS_MEASURE_REG, LL40LS_MSRREG_ACQUIRE },
```

Read:
```
case PHASE_COLLECT: {
        be16_t val;
        // read the high and low byte distance registers
        if (_dev->read_registers(LL40LS_DISTHIGH_REG | LL40LS_AUTO_INCREMENT, (uint8_t*)&val, sizeof(val))) {
            uint16_t _distance_cm = be16toh(val);
            // remove momentary spikes
```

To fix this, I changed the lidar from free-running mode, to single capture.
Then receiving the distance was the same as for v1 modules:
* Send measurement request 
```_dev->write_register(LL40LS_MEASURE_REG, LL40LS_MSRREG_ACQUIRE)```
* Read measurement registers 
```_dev->read_registers(LL40LS_DISTHIGH_REG | LL40LS_AUTO_INCREMENT, (uint8_t*)&val, sizeof(val))```

It now works, I have no idea what has actually changed between the batches.

Please see my [crude bugfix](https://github.com/lukedempsey/ardupilot/commit/1417978e32ac107bf5605935498c5b75a66670cf)

**Version**
Plane 3.9

**Platform**
Plane

**Airframe type**
Quadplane

**Hardware type**
Pixhawk2 Cube, Garmin LidarLiteV3

OXINARF, 2018-10-27:
The new batch you got is likely the new, different version. I just merged #9169, so in a few hours it should be in master builds. It's marked for backport for Plane and Copter, but that's up to vehicle maintainers when to do it.

closed 2018-10-27 by OXINARF

lukedempsey, 2018-10-28:
@OXINARF, I am using the LidarLiteV3, not the LidarLiteV3HP - I saw that pull request but didn't think it would be applicable to me, since I'm using different hardware.

I just ran RFIND_test.cpp off the up-to-date master branch (with your changes), changing the RangeFinder type to ```RangeFinder_TYPE_PLI2CV3``` to work with my Lidar.

Works perfectly! Thanks for your work!