ArduPilot/ardupilot #9652: TFMini driver issue

nikan8, 2018-10-29:
For some reason, Benewake TFMini sensor reports strange values as displayed by sonarrange variable in Mission planner (this has been tested both on FMUv2 and FMUv5). When the sensor switches to long range mode, sonarrange value jumps to about 650 meters. Is there an easy fix for this? Thanks.

rmackay9, 2018-10-30:
yes, the issue is that when the lidar is unable to get a distance it returns a values around 65535 (i.e. -1).  We need to add handling for this in the driver..

rmackay9, 2018-10-30:
I'm hoping to fix this for Copter-3.6.1.

nikan8, 2018-10-30:
Thank you, However, the issue is that when I read the sensor info using TFMini software as well as in PX4 (QGC) I'm able to get a distance estimate. Perhaps, reliability threshold is set too high in Ardupilot tfmini driver? (approximate distance measure when jump to 655 occurs is at approx. 2meters).

closed 2018-11-06 by tridge

nikan8, 2018-11-06:
could this change be backported to arduplane 3.9.2?