ArduPilot/ardupilot #9657: Copter-3.6.0: skips 1st spline waypoint in SITL

rmackay9, 2018-10-30:
in SITL at least (and perhaps in a real vehicle although this hasn't been seen yet), the first Spline waypoint in a mission may be skipped.  To recreate the problem:

1. start master or Copter-3.6 branch in SITL
2. create a mission like below with a takeoff, a spline waypoint and a return-to-launch command
[image]
3. type LOITER to switch to Loiter mode
4. arm throttle (to arm the vehicle)
5. rc 3 1100 (to raise throttle slightly)

you should see that the vehicle climbs to takeoff altitude, the spline waypoint command completes immediately and the vehicle begins the RTL command.

arjun7965, 2018-10-30:
This seems to even happen when spline is the first waypoint after LOITER_TIME. Just recreated this in SITL with:
[image]

SwiftGust, 2018-11-06:
I think not just the commit the link @rmackay9 pointed, also spline time got issue too.
This is SITL - Gazebo result with spline waypoint, it's not only skipping WP 2, also determines very earlier before reaching other WPs 
Note WP6 is just straight WP, other WPs are spline
[image]

SwiftGust, 2018-11-06:
@rmackay9 I made PR for the possible solution of this issue https://github.com/ArduPilot/ardupilot/pull/9710

rmackay9, 2019-03-01:
This is fixed in master and released with Copter-3.6.6.

closed 2019-03-01 by rmackay9