ArduPilot/ardupilot #9672: Rover/Boat: add support for NMEA 2000 echosounders

rmackay9, 2018-11-01:
Rover/Boat currently supports echosounders like the ECT400 that output using the NMEA 0183 protocol but there are many other low cost echosounders out there like the [AirMar DST 800](http://www.airmar.com/uploads/Brochures/dst800.pdf) that use the [NMEA 2000 protocol](https://en.wikipedia.org/wiki/NMEA_2000) so we should add support for these too.

NMEA 2000 is apparently a form of CAN.

tajisoft, 2018-11-12:
@rmackay9 I'm interested in this issue. I want to challenge to implement about NMEA2000. I'll buy a NMEA2000 device to test.

artsiom-sukharau, 2019-07-17:
Hi,
maybe this could be helpful to implement driver https://github.com/ttlappalainen/NMEA2000

therishidesai, 2023-02-07:
Was any work done on this? It looks like the new Torqeedo motors are now CAN only and use NMEA2000 as well so adding support for this would be useful.

rmackay9, 2023-02-07:
@therishidesai,

No developers have started on this I'm afraid.

Apparently the Torqeedo motors with 25kW (or higher) use a different version of CAN called SAE J1939.  I'm happy to add support for these larger Torqeedo motors but I will need access to a motor and that probably means someone would need to send me a motor (even a half broken or used one) or sponsor the purchase of one.  I'm definitely not trying to pressure anyone of course, I'm just putting this out there so if a company comes along and is interested they know who to contact.

rmackay9, 2024-07-02:
An update to this is we have NMEA 2000 support via Lua now in PR https://github.com/ArduPilot/ardupilot/pull/27433.  This doesn't immediately mean that we can use it for GPS, compass and weathervane but it is a step in the right direction perhaps.

tridge, 2024-07-02:
@therishidesai @tajisoft if either of you have a NMEA 2000 echosounder or other NMEA 2000 device and want support then what I would need is a log with LOG_DISARMED=1 using this logging script:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/CAN_logger.lua
with a log from that script and some description of what the correct data should be a driver should be possible

AlksSAV, 2024-07-02:
It would be great to add support for speed log and compass according to the nmea 2000 standard.

chrisdalke, 2025-01-03:
I'm very interested in N2k for ArduRover and can also help provide some datasets from echosounders, GPS, compass, etc. 

That said, NMEA 2000 & J1939 have a fairly expansive set of messages - sensor data, engine and battery telemetry, relay and steering commands, etc. all of which may be very valuable for ArduBoat.

There's a lot of prior art from open source projects which have already done the hard work of decoding all the PGNs, reverse engineering the protocol, implementing address reservation, etc:
- https://github.com/ttlappalainen/NMEA2000
- https://github.com/canboat/canboat

What is the appetite from ArduPilot developers for using the existing NMEA2000 library and hooking it up as one of the CAN protocol options in [AP_CAN](https://github.com/ArduPilot/ardupilot/blob/4360a7ee7aa883bb7e2e1f84a586a0f7cd84cdbb/libraries/AP_CANManager/AP_CAN.h)? Is adding an embedded library like this something that could be considered?

rmackay9, 2025-01-07:
Hi @chrisdalke,

I'm currently working on adding Torqeedo TorqLink support which is also NMEA2K and while I was initially planning on making that C++, I'm now leaning towards just using Lua.  I think we should probably proceed with more Lua drivers (to implement more NMEA2K support) until we find that that becomes too awkward and/or annoying for users.