ArduPilot/ardupilot #9675: Plane/Rover: move read_airspeed into AP_Airspeed library

rmackay9, 2018-11-01:
This issue came up as part of the review of [this PR to add sailboat support to Rover](https://github.com/ArduPilot/ardupilot/pull/9529).

Both Plane and Rover's sensor.cpp files include very similar read_airspeed functions.  We should move these into the AP_Airspeed library.  It would probably make sense to renamed AP_Airspeed::read() to AP_Airspeed::update() and then add the extra bits that the vehicle code does including:

- calling DataFlash::Log_Write_Airspeed
- call AP_Baro::set_external_temperature

WickedShell, 2018-11-01:
@rmackay9 I think we should actually invert the relationship on baro. Rather then airspeed pushing the data, baro should poll the singleton when data it needs data.

EShamaev, 2018-11-09:
@rmackay9 @WickedShell Thank you for rising this as it answered one weird question I had with baro.
In UAVCAN baro, external temperature is set in baro backend AP_Baro_UAVCAN::update, and also it gets rewritten in sensor.cpp with temperature from airspeed sensor.
If you don't mind, I will take a try to handle this raised issue also with fixing the temperatures coming from different sensors as well.

EShamaev, 2018-11-16:
https://github.com/ArduPilot/ardupilot/pull/9744 merged

closed 2018-11-16 by EShamaev