ArduPilot/ardupilot #9677: Without px4io ,how does the pixracer output pwm signal?

whiani, 2018-11-01:
there have output pwm by px4io instead of px4fmu in the RCOutput.cpp file of the libraries/AP_HAL_PX4 directory ,because alt_fd value is -1. so without px4io, how does the pixracer output pwm signal?

OXINARF, 2018-11-01:
The primary output (https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_PX4/RCOutput.cpp#L31) is either the IO or the FMU, the alt output only exists if IO exists.

This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

closed 2018-11-01 by OXINARF

whiani, 2018-11-02:
But _alt_fd is  -1 all the time ,because there ( https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_PX4/RCOutput.cpp#L57 ) is never going to be run. So i think that the primary output is only the IO instead of fmu.
Can the PWM_OUTPUT0_DEVICE_PATH be opened by FMU instead of IO?

whiani, 2018-11-02:
sorry , I have posted it to the forum.