ArduPilot/ardupilot #9678: ROS / Hector SLAM - No EKF2 IMU data to GCS / Wrong FCU time, RTT too high for timesync

bj-neilson, 2018-11-01:
#### Issue details

I think I've installed all as per the ROS and Hector SLAM for Non-GPS Navigation guide.

- Roscore started OK
- RPLidarA2 node started OK
- Hector SLAM and RVIZ started ok.  Lidar seems to be ok in RVIZ.

- Used **roslaunch mavros apm.launch fcu_url:=/dev/ttyUSB1:921600**

I have communication between my laptop that is running ROS and Pixhawk Cube via FTDI usb to serial connection (Telem2).
Mission Planner is running on a separate laptop.  Communicating with Pixhawk via 433Mhz radio (Telem1) 
For example I can arm the FCU from MAVROS by command.  I can set home position on GCS and it is received by MAVROS.  So I seem to have comms in both directions.

I cannot see any EFK2 IMU data coming in.  Any thoughts?  I'd be very grateful for any assistance.

Seeing several warnings/errors:
[ WARN] [1541111777.853076146]: TM: Wrong FCU time.
[ WARN] [1541111790.846931637]: GP: No GPS fix
[ WARN] [1541111799.884152101]: TM : RTT too high for timesync: 1541110387094.33 ms.

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

#### Airframe type

 Mounted on a zero-turn (skid steer) Toro mower

#### Hardware type

Pixhawk Cube

#### Logs
 Output from MAVROS starting and errors.

ben@ben-HP-Pavilion-Laptop-15-cc5xx:~/catkin_ws$ roslaunch mavros apm.launch fcu_url:=/dev/ttyUSB1:921600
... logging to /home/ben/.ros/log/59926cfc-de23-11e8-9b32-28c63fe43a0e/roslaunch-ben-HP-Pavilion-Laptop-15-cc5xx-12293.log
Checking log director
[roslaunch-ben-HP-Pavilion-Laptop-15-cc5xx-12293.log](https://github.com/ArduPilot/ardupilot/files/2540390/roslaunch-ben-HP-Pavilion-Laptop-15-cc5xx-12293.log)
y for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ben-HP-Pavilion-Laptop-15-cc5xx:35097/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
[... 320 lines cut ...]
[ WARN] [1541111535.834575236]: TM: Wrong FCU time.
[ WARN] [1541111545.841609236]: GP: No GPS fix
[ WARN] [1541111559.550263579]: TM : RTT too high for timesync: 1541110387240.68 ms.
[ WARN] [1541111575.861992400]: GP: No GPS fix
[ WARN] [1541111595.872952252]: TM: Wrong FCU time.
[ WARN] [1541111605.870383283]: GP: No GPS fix
[ WARN] [1541111619.590765438]: TM : RTT too high for timesync: 1541110387161.21 ms.
[ WARN] [1541111635.874988455]: GP: No GPS fix
[ WARN] [1541111655.881413129]: TM: Wrong FCU time.
[ WARN] [1541111666.873322959]: GP: No GPS fix
[ WARN] [1541111679.385477288]: TM : RTT too high for timesync: 1541110386835.86 ms.
[ WARN] [1541111697.861154161]: GP: No GPS fix
[ WARN] [1541111716.861513399]: TM: Wrong FCU time.
[ WARN] [1541111728.853716728]: GP: No GPS fix
[ WARN] [1541111739.529722083]: TM : RTT too high for timesync: 1541110386859.63 ms.
[ WARN] [1541111759.850188792]: GP: No GPS fix
[ WARN] [1541111777.853076146]: TM: Wrong FCU time.
[ WARN] [1541111790.846931637]: GP: No GPS fix
[ WARN] [1541111799.884152101]: TM : RTT too high for timesync: 1541110387094.33 ms.
^C[mavros-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

Thanks,
Ben

NickNunno, 2018-11-02:
@rmackay9 this looks to mirror the results myself and others have seen when following the AP Dev docs. I beleive something is missing from the docs, just havent had the time to chase it down.

rmackay9, 2018-11-06:
@bj-neilson,

We should probably do support for this in the Rover forums.. but in any case.. 

- the "TM: RTT too high for timesync" and "TM: Wrong FCU time." messages are expected (for now).  We only have partial support for the TIMESYNC message in AP.  We do account for the lag on the AP side but we do not yet allow syncing the flight controller time to the companion computer.  Still, this shouldn't stop it from working.

Some questions:

1. does the Rover appear on the map?  Hopefully you're using the MP's set-ekf-origin to tell the EKF where the vehicle is in the world and then soon after that it appears on the map?
2. has the GPS been disabled?  I don't think I've seen the "GP: No GPS fix" message on my setup but I could be wrong.  There's some [instructions on the wiki here](http://ardupilot.org/dev/docs/ros-slam.html#configure-ardupilot).
3. how are you checking that the IMU data is not coming into ROS?  Sorry, I'm not very familiar with ROS but I guess you're doing a "rostopic echo xxx" where "xxx" is one of the topics under /mavros?

bj-neilson, 2018-11-06:
@rmackay9 @jmachuca77 @NickNunno 
Hi Randy, thanks for your time.

1. I used "set EKF origin".  That seemed to take a few goes. Maybe it was my mouse location.  Did work though.  I'm sorry I can't recall if it showed the rover on the map.  I'll be trying again later today.

2. Yes GPS disabled as per your excellent guide.

3. I'm a real ROS beginner so am relying heavily on your guide.  Learning on the go.  I was just looking in MP for the Messages your guide mentioned:
EKF2 IMU1 initial pos NED = 0.0,0.0,0.0 (m)
EKF2 IMU1 is using external nav data
EKF2 IMU0 initial pos NED = 0.0,0.0,0.0 (m)
EKF2 IMU0 is using external nav data

I'll try that rostopic echo and see what i can find.

I'm trying a fresh install of ROS/MavROS/MavLink today.  Never know, might find something.  Sticking with Ubuntu 16.04 for now.  I'll let you know how it goes.  Looking forward to your guide continuing!  I'll really be keen to see the next bit you appear to be working on....feeding lidar data back to the FCU for collision avoidance.  Thanks again for the help

bj-neilson, 2018-11-11:
Hi @rmackay9 ,
I did a complete fresh install of Ubuntu 16.04 then followed the ROS Kinetic and MAVROS installation instructions as per their respective wiki’s. Did the GeographicLib install from source as the script install didn’t seem to work well.
Pretty much back where I started.
1. Tried to “set ekf  origin” but had no response to that. No rover appears on MP screen. If I use “set home”, the home location is sent to ROS. I see that being received on the MAVROS window that is running.
2. GPS disabled, all as per your guide. Still shows gps: no gps fix message.
3. As best I can tell from looking through ROS topics, data is flowing both ways between ROS laptop and Cube. 

As an aside, can GPS be used in this case? Asign it a low confidence value in the ekf? Might help correct odometry errors over time and distance?

Thanks for any thoughts.  I’ve still got a bit of wiring to tidy up on the mower and other farm work to do in the meantime. Can then do a bit of gps driving if needed to tune speeds, turning radius etc as you’ve been working on with Kenny Russell etc.

Ta,
Ben

Mohit953, 2020-05-07:
Hi, I am facing the same issue. Can you guide me for the same?

bj-neilson, 2020-05-07:
Hi @Mohit953 ,
I did get it working, but a long time ago now. I’m sorry I don’t remember what was the problem.  Try the Cartographer guide.  Much the same and Cartographer will give you better results anyway.  If you come up with any specific questions please ask.
Ben

siddharthcb, 2021-02-02:
@bj-neilson i know its been long since this thread come up but i am facing exactly same problems as you have mentioned.
my pixhawk is getting harmed but i am getting these problems```
Tried to “set ekf origin” but had no response to that. No rover appears on MP screen. If I use “set home”, the home location is sent to ROS. I see that being received on the MAVROS window that is running.
GPS disabled, all as per your guide. Still shows gps: no gps fix message.
```
do you recall any of the things that you tried?

`Try the Cartographer guide. Much the same and Cartographer will give you better results anyway.` did you get it working after switching to cartographer or hector-slam itself worked for you?

Omiclous, 2023-10-03:
Hi @Mohit953, Have you solved the problem? The same thing is happening to me.