ArduPilot/ardupilot #9682: Autotest for Tradheli giving false positive on arm features test

bnsgeyer, 2018-11-02:
**Issue details**
when autotest is run for tradheli, it fails the arm features test.  In particular it is failing during the specific arming test for tradheli that checks for that the motors interlock does not enable with the aircraft disarmed.  The error given in the autotest log is
AUTOTEST: FAILED: "Arm features": ValueError('Message has no servo8_raw field',)

By the way, this arming feature test should not be done just for tradheli.  This is applicable to anyone using motor interlock

**Version**
Master ArduCopter V3.7-dev (49ce9d02)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
tradheli

**Hardware type**
SITL

**Logs**
[latest autotest log](http://autotest.ardupilot.org/autotest-output.txt)

peterbarker, 2018-11-04:
On Fri, 2 Nov 2018, Bill Geyer wrote:

> when autotest is run for tradheli, it fails the arm features test. In particular it is failing during the specific arming test for tradheli that checks
> for that the motors interlock does not enable [...]

I've pushed in a fix for that....

> By the way, this arming feature test should not be done just for tradheli. This is applicable to anyone using motor interlock

This is quite true.  I don't know why it is guarded.

khancyr, 2018-11-04:
other vehicles don't have interlock feature, so we cannot test it .

bnsgeyer, 2018-11-04:
@khancyr multicopters have the option to use the motor interlock.  Many probably don't but I know @lthall has told me that he uses it on his large quadcopters.  So I would test it for all multirotor frames.

bnsgeyer, 2018-11-04:
@peterbarker thanks for the quick turn around for the fix.

closed 2018-11-04 by bnsgeyer

peterbarker, 2018-11-05:
On Sun, 4 Nov 2018, Pierre Kancir wrote:

> other vehicles don't have interlock feature, so we cannot test it .

RC Option 32.  Applicable to Sub and MultiCopters, too.