ArduPilot/ardupilot #9715: MS4525do sensor not initialized on fmuv5

nikan8, 2018-11-07:
Hello,
MS4525DO airspeed sensor fails to initialize on Pixhawk4 board. The sensor is checked and is in working order (tried on a different firmware and a platform). The sensor is physically plugged into 'I2C A' port on the board. Checked the MAV logs, the lines of code that are triggered are 
ardupilot/libraries/AP_Airspeed/AP_Airspeed.cpp

if (sensor[i] && !sensor[i]->init()) {
  gcs().send_text(MAV_SEVERITY_INFO, "Airspeed[%u] init failed", i);
   delete sensor[i];
   sensor[i] = nullptr;
}

**Platform**
[X ] Plane 3.10.0dev (master)

Thank you for your help.

nikan8, 2018-11-07:
just discovered that 'I2C A' port was excluded from probing in AP_Airspeed/AP_Airspeed_MS4525.cpp
I added `{ 3, MS4525D0_I2C_ADDR },` as another element to addresses[], and then I got the sensor's airspeed readings.

nikan8, 2018-11-30:
can this problem be fixed in current master/latest stable release?

duran9dv, 2019-03-04:
Also trying to connect the 4525 airspeed to Pixhawk 4, running Arudplane latest, it does not read the sensor, what fixed are there here for this ??

duran9dv, 2019-03-04:
> just discovered that 'I2C A' port was excluded from probing in AP_Airspeed/AP_Airspeed_MS4525.cpp
> I added `{ 3, MS4525D0_I2C_ADDR },` as another element to addresses[], and then I got the sensor's airspeed readings.

Is there any possibility you could share this fix with us ?

magicrub, 2019-03-04:
https://github.com/ArduPilot/ardupilot/pull/10697 has been merged

closed 2019-03-04 by magicrub

duran9dv, 2019-03-05:
@nikan8 could you share your settings for the airspeed ? This has been merged into Master from my pull request, but I'm not getting data yet, maybe @magicrub could advise as well ?

nikan8, 2019-03-05:
@duran9dv , can you share your parameters related to airspeed?

nikan8, 2019-03-05:
also, check sensor's physical connections to ph4 board (wire swaps, linkages, etc.). make sure you've uploaded firmware with that modification I suggested.

duran9dv, 2019-03-05:
[image]

duran9dv, 2019-03-05:
which bus should be selected ?

duran9dv, 2019-03-05:
all working! got it, thanks!

bys1123, 2019-03-15:
[image]
So, no need to change this setting after set ARSPD_BUS to 3? @nikan8

ZachDons, 2019-04-11:
@duran9dv Could you post your airspeed parameters? We have the Pixhawk 4 (https://www.sparkfun.com/products/14841) and the MS4525D0 airspeed sensor (https://store.mrobotics.io/mRo-I2C-Airspeed-Sensor-JST-GH-p/mro-classy-arspd-mr.htm) but can not get them to communicate.

bys1123, 2019-04-12:
@ZachDons Are you using master branch?

tylerfaye, 2019-04-12:
@bys1123 I'm working with @ZachDons , we are running the stable version 3.9.7 firmware.

Would using the latest 3.10.0 dev firmware solve this issue? I assumed 3.9.7 included this change because it was released on April 1st while this change was committed to the master branch on March 4th.

tylerfaye, 2019-04-12:
Using a logic analyzer we found that the Pixhawk, upon startup, would only attempt I2C communications inside of notify.init() and AP::compass().init().

Specifically it attempts to write to the NCP5623 and NCP5623_C LED's in the notify section, and then attempts to find all the various I2C compasses after that, but that was all that passed out through the I2C A port.

Between notify.init() and AP::compass().init() we should have also seen it attempting to communicate with the 0x28 address of the MS4525D0 sensor, but it never did attempt to write to or read from that slave address. In our parameters we had set the airspeed sensor to bus 3 using mission planner.

bys1123, 2019-04-12:
@tylerfaye Did you set ARSPD_BUS to 3?

tylerfaye, 2019-04-12:
@bys1123 Yes, the airspeed sensor was set to Bus 3. There was still no communication addressed to the sensor using 0x28, no matter which bus the airspeed sensor was set to in parameters.

It seems like the most likely answer is that the change hasn't been implemented in the stable version yet. At the very least no mention of it is made anywhere in the 3.9.6 or 3.9.7 patch notes. I'll test tomorrow with the latest edition of the firmware.

nikan8, 2019-04-12:
This change is not in 3.9.7.
clone plane3.9 branch, make the change in ms4525 driver (see above) and
recompile

tylerfaye, 2019-04-12:
@nikan8 that's what I figured after looking more closely at the releases.

I'll try the already-built versions available on the ardupilot website first, just because I don't have all the dependencies required to build it on my own machine right now. If that doesn't work I will try my own build though, since then I can make sure the change is present even if the cloned branch wouldn't have it otherwise.

Thanks!

duran9dv, 2019-04-12:
We've been running 3.9.7 and everything is working fine, be sure to check your actual sensor, i spent hours troubleshooting and eventually just swapped out the sensor (two of them!) and it worked with the third.

duran9dv, 2019-04-12:
[image]

WickedShell, 2019-04-12:
`ARSPD_BUS` has no impact on the MS4525 sensor, we scan the buses in a fixed order regardless of the parameter.

duran9dv, 2019-04-12:
juts load the "latest" 3.9.7 and every works great

tylerfaye, 2019-04-12:
Alright, today I was back in the lab and got to test different firmware revisions to see what does and doesn't work for the MS4525D0 sensor and the Pixhawk 4, specifically with the airspeed sensor plugged into the I2C A bus.

This is a long comment since I want to give full detail for anyone else having issues, but I'll include a short summary at the end if people don't care.

1) Stable release 3.9.7 does not work with the Pixhawk 4 and the MS4525D0 airspeed sensor. It does not matter if you download the Arduplane firmware from Mission Planner or if you download it from firmware.ardupilot.org/Plane/stable/Pixhawk4/ - the result is the same each way in that the sensor will always read 0. Changing the parameters does not matter, it will never see the I2C airspeed sensor on the "I2C A" port. The only way to use the MS4525D0 airspeed sensor with a Pixhawk 4 and rev-3.9.7 firmware is to plug the airspeed sensor into the "I2C B" port since Bus 2 (the I2C B bus in Ardupilot) is scanned even in the old version of the code.

2) The beta firmware available both through Mission Planner (clicking the "Beta firmwares" button) and firmware.ardupilot.org/Plane/beta/Pixhawk4/ is functionally identical to the stable 3.9.7 release, in terms of MS4525D0 sensor and Pixhawk 4 compatibility. In fact, if you try to load the beta firmware when the stable firmware is already loaded it will refuse because it believes the firmware is already present. Even if you force the loading of the beta firmware (load "latest" firmware and then load the Beta firmware), it still will only read 0 for the airspeed sensor plugged into the "I2C A" port. 

3) The latest firmware build available at firmware.ardupilot.org/Plane/latest/Pixhawk4/ works properly for a MS4525D0 airspeed sensor plugged into the I2C A port on a Pixhawk 4. The airspeed properly reads and reacts to changes (such as a hair dryer or blowing on the tube). Further testing comfirmed that the parameter ARSPD_BUS doesn't matter, you can set it to anything and the airspeed sensor will still work because the sensor is automatically detected in the code for the MS4525D0.

Photos of the firmware version, working airspeed sensor (blew on it 3 times), and airspeed sensor parameters are below:
[image]
[image]
[image]

I've also found a more elegant possible solution after comparing the MS4525D0 initialization code to the code for the other I2C airspeed sensors. The current code uses the following code snippet while iterating through the "addresses" array with a for-loop to find the sensor:

`_dev = hal.i2c_mgr->getDevice(addresses[i].bus, addresses[i].addr);`

It would be a much cleaner solution to instead use the following code snippet and eliminate the "addresses" array and the for loop altogether:

`_dev = hal.i2c_mgr->getDevice(getBus(), MS4525D0_I2C_ADDR);`

The full content of this proposed change can be see in my pull request #11103

- - - 

# Summary and Conclusion

I would strongly recommend implementing this fix in the next stable version of firmware for Arduplane, whenever that it released, to avoid confusion and allow use of both external I2C ports on the Pixhawk 4 (I2C A and I2C B) for the MS4525D0 sensor. This will make Arduplane more user-friendly and prevent headaches and frustration for anyone using it with a Pixhawk 4 since most I2C airspeed sensors are sold bundled with the cable that connects directly to the pinout/connector style of the I2C A port. There was actually a second team at our school using a Pixhawk 4 for their project that had the same airspeed sensor issues, so it's at least moderately common even if it's not widespread.

In the meantime a small note in the ArduPlane wiki that the MS4525D0 sensors will not work on the I2C A port of the Pixhawk 4 would likely save people a headache like what we had, I'd be willing to put something like that together if necessary or desired.

If you currently have problems with the sensor not working on the I2C A port of your Pixhawk 4, you can either plug it into the I2C B port (if you have a compatible cable) or simply use the "latest" firmware build available at http://firmware.ardupilot.org/Plane/latest/Pixhawk4/ to fix your problem.

CodyGoldman, 2019-04-15:
I am also having this problem with the SDP3X sensor when plugged into I2C A. Updating to the "latest" version of the firmware did not fix the issue with this sensor since it seems like the recent change only applied to the MS4525D0 sensor.

WickedShell, 2019-04-15:
SDP3X relies upon you setting the ARSPD_BUS parameter correctly.

CodyGoldman, 2019-04-15:
I have tried the three possible values for ARSPD_BUS within the range and none seem to do anything. ARSPD_TYPE set to 6, ARSPD_USE to 1, and ARSPD_PIN to 65. Haven't had any luck changing those to find a combination of the three which gets any measurement either.

Plugging in an MS4525D0 sensor with the latest version of the firmware allows a proper reading, but there's still nothing coming in when the SDP3X is connected. I've tried two sensors of the SDP3X model and neither have been able to work.

tylerfaye, 2019-04-15:
@CodyGoldman just to confirm, you have set ARSPD_BUS to 3, right? Because the I2C A port is bus 3, even though the only listed options in Mission Planner are Bus 0, Bus 1, and Bus 2.