ArduPilot/ardupilot #9730: Passthrough support for BLheli_S ESCs not working as expected

Transistorsis, 2018-11-08:
### Bug report

**Issue details**
When connecting to the FC using BLHeliSuite 16, the FC connects, but there is no communication with the ESCs.

Enabling debugging using ESC number 1 as debug, and attempting to read data from an ESC, this is output to the console:

> ESC: OutB(12) 0x37 ack=0x0F
ESC: bad ACK1 0x01
ESC: BL_ReadBuf 8 -> 9
ESC: BL_ConnectEx 0/4 at 0
ESC: cmd_DeviceInitFlash(0)
ESC: BLHeli cmd 0x37 len=1
ESC: OutB(9) 0x31 ack=0x00
ESC: cmd_ProtocolGetVersion
ESC: BLHeli cmd 0x31 len=1
ESC: OutB(12) 0x37 ack=0x0F
ESC: bad ACK1 0x01
ESC: BL_ReadBuf 8 -> 9
ESC: BL_ConnectEx 3/4 at 3
ESC: cmd_DeviceInitFlash(3)
ESC: BLHeli cmd 0x37 len=1
ESC: OutB(9) 0x31 ack=0x00
ESC: cmd_ProtocolGetVersion
ESC: BLHeli cmd 0x31 len=1
ESC: OutB(12) 0x37 ack=0x0F
ESC: bad ACK1 0x01
ESC: BL_ReadBuf 8 -> 9
ESC: BL_ConnectEx 2/4 at 2
ESC: cmd_DeviceInitFlash(2)
ESC: BLHeli cmd 0x37 len=1
ESC: OutB(9) 0x31 ack=0x00
ESC: cmd_ProtocolGetVersion
ESC: BLHeli cmd 0x31 len=1
ESC: OutB(12) 0x37 ack=0x0F
ESC: bad ACK1 0x01
ESC: BL_ReadBuf 8 -> 9
ESC: BL_ConnectEx 1/4 at 1
ESC: cmd_DeviceInitFlash(1)
ESC: BLHeli cmd 0x37 len=1
ESC: OutB(9) 0x31 ack=0x00
ESC: cmd_ProtocolGetVersion
ESC: BLHeli cmd 0x31 len=1
ESC: OutB(12) 0x37 ack=0x0F
ESC: bad ACK1 0x01
ESC: BL_ReadBuf 8 -> 9
ESC: BL_ConnectEx 0/4 at 0
ESC: cmd_DeviceInitFlash(0)
ESC: BLHeli cmd 0x37 len=1
ESC: OutB(9) 0x31 ack=0x00
ESC: cmd_ProtocolGetVersion
ESC: BLHeli cmd 0x31 len=1
ESC: OutB(9) 0x3F ack=0x00
ESC: cmd_InterfaceSetMode(1)
ESC: BLHeli cmd 0x3F len=1
ESC: OutB(10) 0x33 ack=0x00
ESC: cmd_InterfaceGetVersion
ESC: BLHeli cmd 0x33 len=1
ESC: OutB(13) 0x32 ack=0x00
ESC: cmd_InterfaceGetName
ESC: BLHeli cmd 0x32 len=1
ESC: OutB(9) 0x31 ack=0x00
ESC: cmd_ProtocolGetVersion
ESC: BLHeli cmd 0x31 len=1
ESC: escMode=255 portIndex=0 num_motors=4
ESC: MSP_SET_4WAY_IF
ESC: MSP cmd 245 len=0
ESC: MSP_MOTOR
ESC: MSP cmd 104 len=0
ESC: MSP_MOTOR_CONFIG
ESC: MSP cmd 131 len=0
ESC: MSP_MOTOR_3D_CONFIG
ESC: MSP cmd 124 len=0
ESC: MSP_STATUS
ESC: MSP cmd 101 len=0
ESC: MSP_FEATURE_CONFIG
ESC: MSP cmd 36 len=0
ESC: MSP_BUILD_INFO
ESC: MSP cmd 5 len=0
ESC: MSP_BOARD_INFO
ESC: MSP cmd 4 len=0
ESC: MSP_FC_VERSION
ESC: MSP cmd 3 len=0
ESC: MSP_FC_VARIANT
ESC: MSP cmd 2 len=0
ESC: MSP_API_VERSION
ESC: MSP cmd 1 len=0

There are no telemetry wires on these ESCs. But that was not stated as something that would inhibit the use of passthrough.

***Settings***
SERVO_BLH_AUTO: 1 
SERVO_BLH_DEBUG: 1 
SERVO_BLH_MASK: 15 
SERVO_BLH_OTYPE: 0
SERVO_BLH_PORT: 0
SERVO_BLH_TEST: 1
SERVO_BLH_TMOUT: 30
SERVO_BLH_TRATE: 10

**ESCs in use**
spedix ES 30HV 3-6s BLHeli_s

**Version**
Copter 3.6.0 on ChibiOS (2eedcf56)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter 3.6.0
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
Pixracer

**Logs**
No logs were generated in this attempt.

sh83, 2018-11-11:
I could confirm this issue. BLheli_s passthrough does not work.

wiistriker, 2019-01-13:
I faced with this issue too

n1ente, 2019-01-13:
I confirm it still not working with the last beta

sheltercv, 2019-01-26:
I confirm it still not working with the latest.

Bumerangas, 2019-02-25:
Can`t connect too, which version should i use for blheli_s cinfiguration?

flyfisher2, 2019-02-25:
Not working on 3.7 latest for blheli_s

maroquio, 2019-04-12:
Not working yet. Please, devs, take a look at this issue....

maroquio, 2019-04-12:
I'm using HAKRC 4-in-1 15A BLHeli_S and BLHeliSuite 16 can't read the setup. Please, help!

cromartic, 2019-04-23:
I have the same issue...

bys1123, 2019-04-23:
Is BLHeli_32 works?

maroquio, 2019-04-23:
I don't know about BLHeli_32, because I don't have an ESC with this version...

Bumerangas, 2019-05-24:
blhelis_s not woking

Bumerangas, 2019-05-24:
airplane ~2.6 version just working with blheli_s

DeSaaD37, 2020-01-12:
I have bheli32 and having issues connecting. Not sure if this is related or not but will report back here if I figure it out.

Givy, 2021-01-02:
I connected 4 motors to a Holybro PM07 with Pixhawk-4, Ardupilot and 4 motors + TEKKO32 ESCs to use DShot. After a lot of work on the ESCs, Tekko32 by Holybro, I finaly got to see the ESCs in BLHeli32 software. However, it is skipping motor#1 and starts at Motor#2,3,4,5. Can’t even run Motor#5 in BLHeli32, but Motors #2,3,4 respond OK.

Information
Found Multiple ESC configuration:

ESC# 1 : —

ESC# 2 : Airbot_Wraith32_ST - Rev. 32.7 - Multi
[MASTER]
(BLHeli32 Bootloader j)
Input Protocol: DShot
Input Frames Good: 25787 / Bad: 0

ESC# 3 : Airbot_Wraith32_ST - Rev. 32.7 - Multi
[SLAVE]
Setup NOT In SYNC with Master
(BLHeli32 Bootloader j)
Input Protocol: DShot
Input Frames Good: 25798 / Bad: 0

ESC# 4 : Airbot_Wraith32_ST - Rev. 32.7 - Multi
[SLAVE]
Motor Direction: Reversed
(BLHeli32 Bootloader j)
Input Protocol: DShot
Input Frames Good: 25801 / Bad: 0

ESC# 5 : Airbot_Wraith32_ST - Rev. 32.7 - Multi
[SLAVE]
Setup NOT In SYNC with Master
(BLHeli32 Bootloader j)
Input Protocol: DShot
Input Frames Good: 25805 / Bad: 0
OK <
image
image
320×554 15.9 KB

Transistorsis, 2021-01-03:
Seeing as how this issue was about BlHeli_S ESCs, and those are not receiving telemetry support. I am closing this issue.

closed 2021-01-03 by Transistorsis

anbello, 2021-01-18:
This issue has been solved with this merged PR:
https://github.com/ArduPilot/ardupilot/pull/16299

Givy, 2021-01-18:
Were do I download it?

andyp1per, 2021-01-19:
Its also in 4.06rc2 - so you could try that

Givy, 2021-01-19:
Thank you  much.

Givi