ArduPilot/ardupilot #9745: RangeFinder: set RNGFND_DIST_MAX parameter defaults based on driver

rmackay9, 2018-11-10:
We can make the configuration of range finders a little easier for the users by setting the default RNGFND_MAX_CM and RNGFND_MIN_CM values to appropriate values based on the range finder driver.

So for example, the [teraranger one wiki page](http://ardupilot.org/copter/docs/common-teraranger-one-rangefinder.html) states:
- RNGFND_MAX_CM = 1400
- RNGFND_MIN_CM = 20

So we could probably add a couple of lines like this into the AP_RangeFinder_TeraRangerI2C::init() method:

- state.min_distance_cm.set_default(20)
- state.max_distance_cm.set_default(1400)

Divyateja04, 2021-11-01:
I tried to work on this, but I got class `"RangeFinder::RangeFinder_State" has no member "min_distance_cm"` error, do you haev any idea why this might have happened?

hendjoshsr71, 2021-11-01:
@Divyateja04  It is best to link to a branch so that others can see what you are doing.

peterbarker, 2021-11-03:
@Divyateja04 see https://github.com/ArduPilot/ardupilot/pull/19132