ArduPilot/ardupilot #9748: Bug in DataFlash logging for AP_BattMonitor

olliw42, 2018-11-10:
The logging of consumed charge (mah, curr_tot,...) and consumed energy (wh), done here
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_BattMonitor/AP_BattMonitor.cpp#L226-L244
is flawed and incomplete. 

The charge and energy are logged as battery.consumed_mah(battery_instance) and battery.consumed_wh(battery_instance), i.e. in units of mAh and Wh, while CURR_UNITS holds '?' and 'J' = "W.s" for charge (CurrTot) and energy (EnrgTot), respectively.

See
https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/LogFile.cpp#L1397-L1416
and
https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/LogStructure.h#L1138-L1141
and
https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/LogStructure.h#L80-L94

One either could use the already available units { 'i', "A.s" } and { 'J', "W.s" }, and correct the CURR_UNITS field and the DataFlash_Class::Log_Write_Current_instance() function accordingly, or one could introduce new units mAh and Wh, and adapt CURR_UNITS accordingly.

peterbarker, 2018-11-12:
@amilcarlucas Thoughts?

rmackay9, 2018-11-12:
This is probably more for @WickedShell isn't it?

ongspxm, 2019-01-24:
is this still required? I would want to try to work on this, I traced it to `libraries/AP_BattMonitor` and `libraries/AP_Logger`, am I on the right track?

peterbarker, 2019-01-24:
@amilcarlucas SI units only.  display things are what tools are for.

amilcarlucas, 2019-01-25:
Easier to debug, I guess. But yes, SI makes sense.

IamPete1, 2021-07-26:
We now log in ah, https://github.com/ArduPilot/ardupilot/pull/17652 so I think this can be closed.

closed 2021-07-26 by IamPete1