ArduPilot/ardupilot #9769: Aircraft does not hold position on auto takeoff

bnsgeyer, 2018-11-12:
### Bug report

**Issue details**
During takeoff for an auto mission, the aircraft drifts just after takeoff.  When the aircraft reaches 5 m in altitude, it aggressively maneuvers to return to the commanded position.  For heli's it is easy to see this behavior because the tail rotor will cause the aircraft to drift if a right wing down attitude is not achieved just after takeoff.  So this behavior is easy to see.  Since heli and multi's share the same auto takeoff logic, I can only assume that multi's would see the same issue.

Here is a screen shot of an auto takeoff.
[image]
Note the actual position diverges from the commanded position just after takeoff.  At 5 m altitude, the aircraft makes an aggressive right roll up to 30 deg to return to the commanded position.

**Version**
AC 3.6.1 (chibios)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Tradheli

**Hardware type**
Pixhack v5 with Chibios AC 3.6.1
SITL AC 3.6.1

**Logs**
https://drive.google.com/open?id=1LXkTInwSzENUBAWvviEzHJqohJjpHeTc

jschall, 2018-11-14:
You've got WP_NAVALT_MIN set to 5m. Set it to 0.05 and your problem should go away.

bnsgeyer, 2018-11-14:
@jschall yes I just found out that the navalt_min affects when the aircraft begins to hold position. So I guess this is fine for multi's but I still argue that this may not be optimal for heli's, since they immediately begin to drift due to the tailrotor. I am trying to find out why the decision was made to keep the heli from taking up the roll trim attitude on liftoff as is done with all other non auto modes.

bnsgeyer, 2018-11-14:
@tridge  and @rmackay9  I was looking at the Blame for the lines of code in takeoff.cpp in the auto_takeoff_attitude_run routine that keeps the hover roll trim from being applied to the desired roll angle.  Here is the explanation for the commit for this change
"this adds TKOFF_NAV_ALT which controls the altitude above takeoff that
navigation can begin. It is meant for unstable vehicles such as helis
to prevent blade strike during initial takeoff."
I can see why you might want to delay the nav until after takeoff but I don't see a reason to delay the hover roll trim from taking effect.  It is initiated when the land_complete flag is false.  

I would like to understand what reasoning went behind this decision before I decide if I want to remove the line keeping the aircraft from attaining the hover roll trim on takeoff.

bnsgeyer, 2019-06-22:
The line of code suppressing the hover roll trim on takeoff was removed.  See PRs #11164 and #10935.  Closing issue.

closed 2019-06-22 by bnsgeyer