ArduPilot/ardupilot #9774: SITL octa-quad failed to flight after autotune

khancyr, 2018-11-13:
### Bug report

After doing an autune in SITL for the default octa-quad, it cannot takeoff anymore.
To reproduce : 
takeoff in loiter
switch to autotune
wait completion
land
get back to stabilize
arm throttle
rc 3 1700 

The copter won't pass 3m and complain of thrust lose.

_Please describe the problem_

**Version**
Confirmed on https://github.com/ArduPilot/ardupilot/commit/d824cf0a4086f83ed43e9d460f42cb766906f11d

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
octa-quad . I didn't test on other frames

KimHyungSub, 2021-06-01:
I tested this bug on 0bb18a153c. It seems that the bug is fixed.

rmackay9, 2021-06-01:
OK great thanks.  I think I might close this one.  @khancyr feel free to re-open if I've made a mistake closing this.

closed 2021-06-01 by rmackay9