ArduPilot/ardupilot #9776: RangeFinder: add support for VL53L1X

rmackay9, 2018-11-14:
This enhancement request is to add support for the low-cost 4m range ST VL53L1X lidar. This can be found from various retailers including [sparkfun](https://www.sparkfun.com/products/14722)

muramura, 2018-11-14:
https://www.pololu.com/product/3415/specs
I know Pololu's device.

nghiajenius-dev, 2018-11-15:
Hi,

I wrote a driver for this sensor to use it with my 250mm quadcopter, based on Pololu code: [(https://github.com/pololu/vl53l1x-arduino)]. I will do some clean up and make a PR this weekend.

nghiajenius-dev, 2018-11-17:
Hi @rmackay9,
I have just made a PR for this sensor here: https://github.com/ArduPilot/ardupilot/pull/9845

avncalst, 2019-01-20:
I have included the VL53L1X ToF in a python-dronekit script on the RPi3 companion computer of my drone. The script also includes artificial intelligence proximity sensing. Code to be found on GitHub https://github.com/avncalst/drone_cnn. Works great.

chobitsfan, 2019-05-24:
I have fly current master (#9845 merged) with katute f7 aio and vl53l1x. Rangefinder reading looks good when copter is disarmed and I hold the copter to move it up and down. But in real flight, ardupilot stop get reading after copter climbing up to 0.5m. 

[image]

[vl53l1x_flight_log.zip](https://github.com/ArduPilot/ardupilot/files/3215402/vl53l1x_flight_log.zip)

chobitsfan, 2019-05-25:
After remove black soft pads on the floor, everything works fine.  It seems that the material and black color absorbed laser from vl53l1x.

hi @rmackay9 I have tested it on katute f7 on a Tarot 140 FPV frame. It fly well indoor in althold mode.

OXINARF, 2019-05-25:
This is supported since #9845 was merged. Thanks to @nghiajenius-dev!

closed 2019-05-25 by OXINARF