ArduPilot/ardupilot #9780: PXFmini MPU9250 INS and Compass same angle?????

karu2003, 2018-11-14:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/board/linux.h---
define HAL_INS_DEFAULT_ROTATION ROTATION_YAW_270
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass.h
elif CONFIG_HAL_BOARD == HAL_BOARD_LINUX && (CONFIG_HAL_BOARD_SUBTYPE == HAL_BOARD_SUBTYPE_LINUX_ERLEBRAIN2 || \
  CONFIG_HAL_BOARD_SUBTYPE == HAL_BOARD_SUBTYPE_LINUX_PXFMINI)
define MAG_BOARD_ORIENTATION ROTATION_YAW_270
http://www.invensense.com/wp-content/uploads/2015/02/PS-MPU-9250A-01-v1.1.pdf
Page 38 
How are you doing? INS and compass have the same angle of displacement. According to the documentation, the INS and Compass are shifted.
Andrew.

vierfuffzig, 2018-11-14:
i'm not firm with linux boards, but we needed to add a mag rot define for MPU9250 on F35Lightning too:
https://github.com/ArduPilot/ardupilot/pull/9468

IamPete1, 2021-08-29:
Since the PXFmini is no longer available I'm going to close this.

closed 2021-08-29 by IamPete1