ArduPilot/ardupilot #9782: Allow channel 6 tuning to change loiter speed

sher1ock, 2018-11-14:
### Feature request

**Is your feature request related to a problem? Please describe.**
There isn't an easy way to change between speeds while in loiter.
**Describe the solution you'd like**
Add LOIT_SPEED to the channel 6 tuning.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
I know that at some point it was added to the menu but didn't actually work and later got removed. This was years ago.

rmackay9, 2020-01-15:
This request was repeated again here on the forums: https://discuss.ardupilot.org/t/loiter-speed-adjustment/51071

rmackay9, 2025-02-10:
Took 7 years but we got there in the end!

PR #14188 (2025-02-10): Copter: Allow CH6 tuning to change loiter speed
This has been requested a few times in the past as well as in #9782.
This has been tested on SITL(incremented channel 6 by 100 every few seconds), and works well: 
[image]
Can't test it on actual copter due to lockdown.
Looking forward to any improvements that can be suggested! 

Fixes #9782
closed 2025-02-10 by rmackay9