ArduPilot/ardupilot #9792: In LOITER descending by pitch-stick input not possible as described in Wiki

lorbass, 2018-11-15:
#### Issue details
When I want to descend in Loiter Mode by Stick input of Pitch the Wing descend but proportionally to
the difference to the original altitude the THR_OUT increase up to THR_MAX and of course also the speed.
Even when reducing with Throttle Stick down to min. And after releasing the Pitch Stick it climbs
immediately on the original altitude.
In the log the param TECS dhdem remains on its Loiter start value and the Pitch Intput has no
influence on it. As it does in FBWB Mode.
I did not test climbing in Loiter, may be even worse (Throttle reduced to Zero?

[image]

#### Version
Plane 3.8.5
#### Platform

[ VTOL] Plane

#### Airframe type
Tailsitter, vectored

#### Hardware type
Pixhawk
#### Logs
https://drive.google.com/open?id=1WBTw4EdCpFqh4jdZDQ2qHQTw_2AeQihT
https://youtu.be/C8AdftuPBG0

IamPete1, 2018-11-15:
I think its behaving as it should. There is a parameter that disables pitch and roll manual override that removes your ability to give pitch inputs ([STICK_MIXING](http://ardupilot.org/plane/docs/parameters.html#stick-mixing-stick-mixing)). So as it is its trying to maintain altitude while your changing the pitch so the throttle is doing the right thing. 

Your suggesting that if you give it a pitch input it controller the altitude rather than the pitch, this would move your loiter up and down as in FBWB? Perhaps the solution is to add a FBWB option to the stick mixing parameter.

IamPete1, 2022-05-22:
Loiter can now be controlled like FBWB using the `FLIGHT_OPTIONS` bit `12: Enable FBWB style loiter altitude control`

closed 2022-05-22 by IamPete1