ArduPilot/ardupilot #9796: Sub: Add support for setting individual motor pwm directly via MAVLINK

patrickelectric, 2018-11-15:
Moved from: bluerobotics/ardusub#166
Original author: @jaxxzer

### Feature request

**Platform**
[ ] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[x] Submarine

slovak194, 2019-07-12:
Hi, @patrickelectric! Do you know, if any work were done in this direction?
Thanks!

patrickelectric, 2019-07-12:
I believe that was done, but not shared. You can try to do some search in https://discuss.bluerobotics.com.

jaxxzer, 2019-07-12:
Nothing new here AFAIK. It will require changes to https://github.com/mavlink/mavlink.

slovak194, 2019-07-12:
Thanks, will take a look at the forum more.

tsaoyu, 2020-01-15:
Any progress on this?

Obert94, 2020-03-27:
Any Updates on the individual thruster control?

josh-holder, 2021-10-14:
Has this been fixed/ has anyone found a workaround for this? Tried using the advice given [here](https://discuss.bluerobotics.com/t/how-to-control-thrusters-independently/9870/6) but the thrusters are still not responding. Not sure if I'm doing something wrong or it still is not supported.

ES-Alexander, 2021-11-02:
[Latest progress](https://discuss.bluerobotics.com/t/how-to-control-thrusters-independently/9870/11) includes a hacky (and somewhat ill-advised/potentially dangerous) workaround, but as @jaxxzer mentioned above, a proper solution for this will require a suitable message to be implemented in mavlink.