ArduPilot/ardupilot #9804: Sub: Rate limit motors test messages

patrickelectric, 2018-11-15:
Moved from: bluerobotics/ardusub#162
Original author: @jaxxzer

### Feature request
Somehow, it is possible to get this output in an instant:

```
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
Got MAVLink msg: COMMAND_ACK {command : 209, result : 4}
APM: Arm motors before testing motors.
APM: motor test initialization failed!
```

**Platform**
[ ] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[x] Submarine

jaylthegreat, 2019-07-15:
To make this more clear, you can get this (or at least a similar) output by navigating to the motor  setup menu in QGC and then either clicking the switch to start the motor test, or by arming it with a configured button.

IMPORTANT NOTE:
This may be more of a QGC issue, but if you leave that menu and go back the main flight view, you will get the same error output when you arm the motors.

These observations are purely from SITL

jaxxzer, 2019-07-15:
@jaylthegreat You have encountered this: https://github.com/mavlink/qgroundcontrol/issues/7275#issuecomment-480306112

Williangalvani, 2019-09-11:
The issue is that QGC is able to get into a state where the motor test is not running in ardusub, but it tries to send motor test messages anyway. 

For example, if you stop the motor test and start again without waiting for the cooldown, the button will be able to arm the motors and send test messages, but ardusub will refuse to move them and reply with a Failure, which results in this:
[image]
[image]

This will be partially fixed by https://github.com/mavlink/qgroundcontrol/pull/7808

closed 2019-09-27 by jaxxzer