ArduPilot/ardupilot #9807: Sub: Consider T100/T200 modelling in software

patrickelectric, 2018-11-15:
Moved from: bluerobotics/ardusub#159
Original author: @jaxxzer

### Feature request
This can help with output current predictions and limiting to prevent transient spikes.

**Platform**
[ ] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[x] Submarine

slovak194, 2019-07-05:
I have some work done for the T200 system identification. Does any help needed?

patrickelectric, 2019-07-05:
Hi @slovak194 can you share your work ?
Right now we have the V/I/kgf plots but we are missing model information (frequency or time domain), do you have this kind of information ?

slovak194, 2019-07-05:
@patrickelectric, I've done an initial system ID [here](https://colab.research.google.com/drive/11KaqoXNNUMft9kXzFgZMYtdeAh3p9T2U) and hope to continue in this direction soon.
Could you please, share V/I/kgf plots as well? I can't find it on blue-robotics web site.
If you have more experimental data for T200 and for the robot itself, please, share it. It will help people to understand how to control your robot properly.

patrickelectric, 2019-07-07:
Hi @slovak194, you can check the [charts here](https://plot.ly/~rjehangir), it's possible also to fetch the CSV raw data from plot.ly website. 
Unfortunately I don't have further information to share besides the one in our website. But, I'll try to update you if any other relevant information comes up.

But, I believe that your model is good enough to help with this particular issue here.

slovak194, 2019-07-07:
Thanks for sharing, @patrickelectric!
If you have any additional info regarding modeling of a robot itself - please, let me know. BTW, what kind of models are used in sitl simulation?

patrickelectric, 2019-07-08:
Hi @slovak194,we did some simple approximations for the SITL model, you can [check it here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_Submarine.h#L69).

As you can see, the buoyancy force/volume is calculated with our knowledge of the new buoyancy from the original vehicle, and `equivalent_sphere_area` is calculated with a sphere radius that is empiric, we changed the value until it feels like the real vehicle.

slovak194, 2019-07-08:
Yeah, I'm already looking into it. BTW, can you also share working SITL setup as you do that a lot? I've found in few threads on your forum that there is a "proper" set of parameters has to be applied when doing sitl. But param files are pretty outdated.
I did fast try without success, manual mode behaves somehow unexpectedly. The sub character is doubling and moving sideways regardless of controls I apply to it.

slovak194, 2019-07-08:
PS: this issue has a chance to become a useful collection of links :+1:

patrickelectric, 2019-07-08:
For sitl, please [follow ardupilot documentation](http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html).

slovak194, 2019-07-18:
> Hi @slovak194, you can check the [charts here](https://plot.ly/~rjehangir), it's possible also to fetch the CSV raw data from plot.ly website.

Hi, @patrickelectric!

I'm trying to use data you have pointed me to. But it is a bit unclear what is the correspondence between columns.

Here is the list of columns in csv file:

**"thrust @ 16V"**
"rpm"
"power 16V"
"current 16V"
"voltage"
**"pwm"**
**"thrust (kgf) 12V"**
"efficiency (g/W)"
"Col9"
"Col10"
**"thrust 12V"**
"rpm #2"
"power 12V"
"current 12V"
"voltage #2"
**"pwm #2"**
**"thrust (kgf) 12V #2"**
"efficiency (g/W) #2"

I'm particularly interested in thrust = fcn(pwm) relationship.
Could you please clarify which PWM column corresponds to which thrust.

Many thanks!