ArduPilot/ardupilot #9808: Sub: Rework joystick input and pilot input in general

patrickelectric, 2018-11-15:
Moved from: bluerobotics/ardusub#158
Original author: @jaxxzer

### Feature request
We need some sort of input abstraction layer that rc input and mavlink/joystick input sit on top of.

**Platform**
[ ] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[x] Submarine

jaxxzer, 2019-01-03:
Our inputs are limited to 4 analog axes and 16 digital buttons, this is very constraining. I would like to support an arbitrary number of analog/digital inputs with flexible assignments to things like inputs, input trims, lights channels 1, 2, 3... multiple camera gimbals, multiple manipulator joints etc.

ES-Alexander, 2021-12-20:
[Current `MANUAL_CONTROL` motion handling](https://github.com/ArduPilot/ardupilot/blob/master/ArduSub/joystick.cpp#L53) works by converting to RC overrides. This would likely be handled better by a more general abstraction (as above), but if the RC overrides conversion is kept it should at least be updated to work off parameter limits (e.g. [`RCn_MIN`](https://www.ardusub.com/developers/full-parameter-list.html#rcnmin-rc-min-pwm)/[`MAX`](https://www.ardusub.com/developers/full-parameter-list.html#rcnmax-rc-max-pwm)) rather than hardcoded 1100-1900 values.

This still has the issue that if a user remaps the RC channels manual control ceases to work as expected, but that seems infeasible to check for at every manual control command, and janky to update a table on update of those parameters, so is likely more of a reason for creating an underlying abstraction that can be used by all interfaces rather than another thing to patch.