ArduPilot/ardupilot #9812: Sub: Allow setting attitude/depth targets via MAVLink

patrickelectric, 2018-11-15:
Moved from: bluerobotics/ardusub#146
Original author: @jaxxzer

### Feature request
Currently this is only available in GPS-enabled modes. We should allow these commands to be sent at any arbitrary point and execute them whenever possible.

**Platform**
[ ] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[x] Submarine

slovak194, 2019-07-05:
Hi! Are there any news or info regarding this issue? I was about to implement something like that by my own. Does work on that already in progress?

Thanks!

patrickelectric, 2019-07-05:
Hi @slovak194 	It's possible to set attitude in `alt_hold` mode, you can check the [PR here](https://github.com/ArduPilot/ardupilot/pull/8322).

IIRC, altitude is not possible yet with `set_position_target_*` without a GPS, you can read more about [noGPS mode here](http://ardupilot.org/copter/docs/ac2_guidedmode.html#guided-nogps). You can try to create a PR to add a new message for that, or you can send a fake GPS position and use the normal functions. With the fake GPS, you'll need to replicate Z with the barometer alt estimation.

slovak194, 2019-07-05:
Hi, @patrickelectric! 

>> It's possible to set attitude in alt_hold mode, you can check the PR here.
Many thanks, I'll take a look. I'll kindly ask you if there are questions.

jaxxzer, 2019-07-09:
We still need to add depth control and handling inside of stabilize mode.

closed 2019-09-27 by jaxxzer