ArduPilot/ardupilot #9830: Sub: servo output channels do not respect configured min/max limits when function is set to rc passthrough

patrickelectric, 2018-11-15:
Moved from: bluerobotics/ardusub#120
Original author: @jaxxzer

### Feature request

**Platform**
[ ] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[x] Submarine

JonPompa, 2020-07-31:
Does this look like the root of the issue? 
In Ardupilot/librariesAP_Motors/AP_Motors6DOF.cpp, the pwm levels have hard coded scale and center numbers.

```
int16_t AP_Motors6DOF::calc_thrust_to_pwm(float thrust_in) const
{
    return constrain_int16(1500 + thrust_in * 400, _throttle_radio_min, _throttle_radio_max);
}

void AP_Motors6DOF::output_to_motors()
{
    int8_t i;
    int16_t motor_out[AP_MOTORS_MAX_NUM_MOTORS];    // final pwm values sent to the motor

    switch (_spool_state) {
    case SpoolState::SHUT_DOWN:
        // sends minimum values out to the motors
        // set motor output based on thrust requests
        for (i=0; i<AP_MOTORS_MAX_NUM_MOTORS; i++) {
            if (motor_enabled[i]) {
                motor_out[i] = 1500;
            }
        }
        break;

```

IamPete1, 2021-09-04:
Duplicate of https://github.com/ArduPilot/ardupilot/issues/18379

closed 2021-09-04 by IamPete1