ArduPilot/ardupilot #9835: Sub: Make sure pilot input gain does not affect position controller outputs

patrickelectric, 2018-11-15:
Moved from: bluerobotics/ardusub#84
Original author: @jaxxzer

### Feature request
Verify report that position hold performance depends on input gain setting.

**Platform**
[ ] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[x] Submarine

siddheshih, 2018-11-20:
This is my first issue , how do I go about solving it?

jaxxzer, 2018-11-20:
@siddheshih Thank you for your interest and contribution, I do not believe that this is a good 'first issue'. You may either inspect/step through the source code to verify that the logic is correct. Or you may use software in the loop to run a simulation with different gain settings and look at any difference in the data.

Williangalvani, 2024-02-23:
closing this as position hold has been reworked quite a bit and is being used for a while now with no issues. 
Feel free to re-open if the issue arrises again.

closed 2024-02-23 by Williangalvani